esphome/issues #7079: 2025.5 version problem with stereo audio

boskikak, 2025-05-28:
### The problem

Testing on 3 different ESP32 S3 with IDF framework. Only one channel plays each time.  Downgrade to an older version eliminates this problem. It's probably connected z #7077 

### Which version of ESPHome has the issue?

2025.5

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP32-IDF

### Board

ESP32 S3 N16R8

### YAML Config

```yaml
i2s_audio:
  - id: i2s_output
    i2s_lrclk_pin: ${pin_lrclk_spk}
    i2s_bclk_pin: ${pin_bclk_spk}
  - id: i2s_input
    i2s_lrclk_pin: ${pin_lrclk_mic}
    i2s_bclk_pin: ${pin_bclk_mic}

micro_wake_word:
  id: mww
  models:
    - model: alexa
      probability_cutoff: 0.5
      sliding_window_size: 2
  vad:
  microphone: mic
  on_wake_word_detected:
      if:
        condition:
            - media_player.is_playing:
        then:
         - logger.log: "S T O P    M U S I C"
         - switch.turn_on: active_music
         - delay: 100ms 
         - switch.turn_off: led_clock
         - script.stop: detecting
         - delay: 100ms 
         - media_player.pause       
         - voice_assistant.start:
            wake_word: !lambda return wake_word;
         - script.execute: detecting
         - delay: 50ms   
         - rtttl.play: 'two_short:d=4,o=5,b=100:16e6,16e6'
        else:
         - switch.turn_off: active_music
         - switch.turn_off: led_clock
         - delay: 100ms 
         - script.stop: detecting
         - delay: 200ms 
         - voice_assistant.start:
            wake_word: !lambda return wake_word;
         - script.execute: detecting
         - delay: 50ms   
         - rtttl.play: 'two_short:d=4,o=5,b=100:16e6,16e6'

voice_assistant:
  id: va
  microphone: mic
  media_player: player
  use_wake_word: false
  noise_suppression_level: 0
  volume_multiplier: 1
  auto_gain: 0dBFS 
  on_error:
    - delay: 100ms
    - script.execute: reset_ww
    - script.execute: error
    - script.stop: detecting
  on_tts_start:
    - switch.turn_off: led_clock
    - script.stop: detecting
    - media_player.volume_set: !lambda return id(vol_assist).state; 
    - delay: 100ms
    - script.execute: speaking
  on_start:
    - logger.log: "M I C R O P H O N E    S T A R T"
  on_end:
      if:
        condition:
            - switch.is_on: active_music
        then:
          - light.turn_off: ring
          - switch.turn_off: led_clock
          - script.execute: reset_ww
          - wait_until:
               condition:
                 not:
                   media_player.is_announcing:
          - delay: 1s
          - media_player.play  
          - logger.log: "E N D    A S S I S T A N T   P L A Y    M U S I C"
        else:      
          - light.turn_off: ring
          - switch.turn_off: led_clock
          - delay: 200ms
          - script.execute: reset_ww
          - logger.log: "E N D    A S S I S T A N T   W I T H O U T    M U S I C"

media_player:
  - platform: speaker
    id: player
    name: "Speaker"
    internal: False
    icon: mdi:dots-hexagon
    task_stack_in_psram: true
    volume_increment: ${vol_step}
    volume_min: 0.0
    volume_max: 0.96
    media_pipeline:
      speaker: media_spk_resampling_input
      format: FLAC     # FLAC is the least processor intensive codec
      num_channels: 2
      sample_rate: 48000
    announcement_pipeline:
      speaker: announcement_spk_resampling_input
      format: FLAC     # FLAC is the least processor intensive codec
      num_channels: 1  # Stereo audio is unnecessary for announcements
      sample_rate: 48000
    on_state:
      then:
        if:
          condition:
            and:
             - media_player.is_announcing:
             - not:
                 voice_assistant.is_running
          then:
           - logger.log: "S T A R T    T T S"
           - switch.turn_off: led_clock
           - delay: 500ms
           - script.execute: speaking
           - delay: 1s
           - wait_until:
               condition:
                 not:
                   media_player.is_announcing:
           - script.execute: reset_ww
           - light.turn_off:  ring 
           - media_player.volume_set: !lambda return (id(vol_music).state);
           - logger.log: "E N D   T T S"
           - delay: 4s
           - switch.turn_on: led_clock
          else:
            if:
              condition:
                - media_player.is_announcing:
              then:
                - logger.log: "S T A R T    A S S I S T"
                - wait_until:
                    condition:
                      not:
                        media_player.is_announcing:
                - logger.log: "E N D    A S S I S T"
                - script.execute: reset_ww
                - light.turn_off:  ring 
                - media_player.volume_set: !lambda return (id(vol_music).state);
                - delay: 4s
                - switch.turn_on: led_clock

speaker:
  - platform: i2s_audio
    id: i2s_audio_speaker
    sample_rate: 48000
    i2s_mode: primary
    i2s_dout_pin: ${pin_din_spk}
    bits_per_sample: 32bit
    i2s_audio_id: i2s_output
    dac_type: external
    channel: stereo
    timeout: never
    buffer_duration: 100ms
  - platform: mixer
    id: mixer_speaker_id
    output_speaker: i2s_audio_speaker
    source_speakers:
      - id: announcement_spk_mixer_input
      - id: media_spk_mixer_input
  - platform: resampler
    id: media_spk_resampling_input
    output_speaker: media_spk_mixer_input
  - platform: resampler
    id: announcement_spk_resampling_input
    output_speaker: announcement_spk_mixer_input

microphone:
  - platform: i2s_audio
    id: mic
    adc_type: external
    i2s_din_pin: ${pin_din_mic}
    channel: left
    i2s_audio_id: i2s_input
```

### Anything in the logs that might be useful for us?

```txt

```

Bascht74, 2025-06-01:
Could you try adding: 

external_components:
  - source:
      type: git
      url: https://github.com/esphome/esphome
      ref: 2025.4.2
    components:
      - logger           # changed 5 days ago

clean the build files and compile again?
Maybe it is releated to 
https://github.com/Blackymas/NSPanel_HA_Blueprint/issues/2543#issuecomment-2927463745
(long start time with the new logger, chrashes with the new logger, maybe it has an influence here, too)

This is just a shot in the dark...

boskikak, 2025-06-15:
> Could you try adding:
> 
> external_components:
> 
> * source:
>   type: git
>   url: https://github.com/esphome/esphome
>   ref: 2025.4.2
>   components:
>   
>   * logger           # changed 5 days ago

Unfortunately it didn't help.

boskikak, 2025-06-29:
TTS & audio playing still work on one channel