esphome/issues #7095: athom smart plug v2 unable to update to 2026.5.1 and 2025.5.2

dnedeltchev, 2025-06-04:
### The problem

I have 2 x athom smart plug v2 and I've not been able to get them to update to 2026.5.1 and now 2025.5.2. I get the following error.

[image]

### Which version of ESPHome has the issue?

2025.5.1 and 2025.5.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2025.5.3

### What platform are you using?

ESP8266

### Board

Odroid n2 plus

### YAML Config

```yaml

```

### Anything in the logs that might be useful for us?

```txt

```

pkejval, 2025-06-04:
Same issue there (obviously). 2025.5.0 is last version that "fitted" my plug with OTA update.

npa-homeassistant, 2025-06-09:
FWIW, mine are failing in the same way. Hopefully somone will be able to streamline something to get then working (updating) again.

dbensmith, 2025-06-12:
If you don't need the web server hosted by the plug, you can edit the `web_server` component of your device YAML to remove it:
```
web_server: !remove
```
This saves ~160 KB of flash space which was enough for my Athom plugs to accept the 2025.5.2 update OTA.

I like having the web server as it's a safeguard against any connectivity issues with HA. But I can live without it for now.

Cheesebaron, 2025-06-18:
What I ended up doing is as per instructions mentioned her: https://github.com/athom-tech/athom-configs/issues/122#issuecomment-2929332266

I added the platformio options under the `esphome` node so my config looks like:

```yaml
esphome:
  name: ${name}
  name_add_mac_suffix: false
  friendly_name: ${friendly_name}
  platformio_options:
    board_upload.flash_size: 2MB
    board_upload.maximum_size: 2097152
    board_build.ldscript: eagle.flash.2m.ld
```

I also commented out the encryption and key nodes

Then I flashed with this configuration. After than I removed the comments for encryption and key nodes and flashed again. Since now there should be enough space on the device.

Had to do this for 5 devices, which wasn't too bad.

However, another issue seems to be that `relay_restore_mode: RESTORE_DEFAULT_ON` is not respected anymore and when the plug reboots after flashing, it turns off.