esphome/issues #7099: Regression: Custom component C++ inheritance from `UptimeSecondsSensor` broken since 2025.2.0 `uptime` refactor unless dummy `uptime` sensor is added to YAML

echavet, 2025-06-05:
### The problem

**Problem Description:**
Since ESPHome version 2025.2.0, which included a refactoring of the `uptime` component (moving `UptimeSecondsSensor` into a `sensor/` subdirectory), custom components that directly inherit in C++ from `esphome::uptime::UptimeSecondsSensor` fail to build (likely due to linker errors for missing `UptimeSecondsSensor` method implementations).

This issue occurs even if the custom component's Python configuration specifies `AUTO_LOAD = ["uptime"]` and `DEPENDENCIES = ["uptime"]`.

The build succeeds if a dummy `sensor: - platform: uptime` is explicitly added to the user's YAML configuration, forcing the compilation and linking of the `UptimeSecondsSensor` C++ sources.

**Context:**
- ESPHome Version: 2025.2.0 and later (problem started here)
- Affected Component: Custom components using `UptimeSecondsSensor` as a C++ base class. Example: [Link to your MitsubishiCN105ESPHome component's relevant code, e.g., HpUpTimeConnectionSensor.h]

**To Reproduce:**

1.  **Create a minimal custom component (`my_custom_sensor`)**:
    * `my_custom_sensor.h`:
        ```cpp
        #pragma once
        #include "esphome/core/component.h"
        #include "esphome/components/sensor/sensor.h"
        // Attempt to include the refactored path
        #include "esphome/components/uptime/sensor/uptime_seconds_sensor.h" 

        namespace esphome {
        namespace my_custom_sensor {

        class MyUptimeDerivedSensor : public uptime::UptimeSecondsSensor { // Inherit
         public:
          void setup() override {
            // Minimal setup, or call a base method if needed for testing
            UptimeSecondsSensor::setup(); 
          }
          void update() override {
            // Minimal update, or call a base method
            UptimeSecondsSensor::update();
            ESP_LOGD("custom", "MyUptimeDerivedSensor update called.");
            publish_state(this->uptime_); // uptime_ is a protected member of UptimeSecondsSensor
          }
          void dump_config() override {
            LOG_SENSOR("", "My Uptime Derived Sensor", this);
          }
          std::string unique_id() override { return "my_uptime_derived"; }
        };
        } // namespace my_custom_sensor
        } // namespace esphome
        ```
    * `sensor.py` (for the custom component):
        ```python
        import esphome.codegen as cg
        import esphome.config_validation as cv
        from esphome.components import sensor
        from esphome.const import CONF_ID

        # This custom component will AUTO_LOAD uptime
        AUTO_LOAD = ['sensor', 'uptime'] 
        DEPENDENCIES = ['uptime']

        my_custom_sensor_ns = cg.esphome_ns.namespace('my_custom_sensor')
        MyUptimeDerivedSensor = my_custom_sensor_ns.class_(
            'MyUptimeDerivedSensor', sensor.Sensor, cg.PollingComponent
        )

        CONFIG_SCHEMA = sensor.sensor_schema(
            MyUptimeDerivedSensor,
            accuracy_decimals=0
        ).extend(cv.polling_component_schema('60s'))

        async def to_code(config):
            var = cg.new_Pvariable(config[CONF_ID])
            await cg.register_component(var, config)
            await sensor.register_sensor(var, config)
        ```

2.  **Create a minimal YAML configuration using this custom component WITHOUT an explicit `uptime` sensor:**
    ```yaml
    esphome:
      name: test_uptime_issue

    esp32: # or esp8266
      board: esp32dev
      framework:
        type: esp-idf # or arduino

    logger:

    external_components:
      - source: # local path to your my_custom_sensor component
          type: local
          path: components/my_custom_sensor 

    sensor:
      - platform: my_custom_sensor # Using the custom component
        name: "My Derived Uptime Sensor"
    ```

3.  **Attempt to compile.** It should fail with linker errors related to `esphome::uptime::UptimeSecondsSensor` methods.

4.  **Modify the YAML to include a dummy `uptime` sensor:**
    ```yaml
    # ... (rest of the YAML as above) ...
    sensor:
      - platform: my_custom_sensor
        name: "My Derived Uptime Sensor"
      - platform: uptime # Add this
        name: "Dummy Uptime For Build" 
    ```

5.  **Clean build files and recompile.** The build should now succeed.

**Expected Behavior:**
Specifying `AUTO_LOAD = ["uptime"]` and/or `DEPENDENCIES = ["uptime"]` in a custom component's Python definition should be sufficient to ensure that all necessary C++ code (both headers for compilation and source files for linking) for classes within the `uptime` component (like `UptimeSecondsSensor`) are included in the build if those C++ classes are used as base classes by the custom component. The refactoring of `uptime` seems to have broken this expectation for C++ source file linkage.

**Impact:**
Custom components relying on direct C++ inheritance from `UptimeSecondsSensor` (and potentially other refactored component classes) are now harder to use and require users to add potentially unnecessary dummy entities to their YAML to ensure a successful build. This is a regression in developer experience for custom component creators and usability for users of such components.

**Workaround:**
The current workaround is to ask users to manually add an `uptime` sensor (e.g., `sensor: - platform: uptime`) to their YAML configuration.

Thank you for looking into this.

### Which version of ESPHome has the issue?

2025.2.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2025.*

### What platform are you using?

ESP32

### Board

all

### Component causing the issue

uptime

### YAML Config

```yaml
esphome:
  name: test_uptime_issue

esp32: # or esp8266
  board: esp32dev
  framework:
    type: esp-idf # or arduino

logger:

external_components:
  - source: # local path to your my_custom_sensor component
      type: local
      path: components/my_custom_sensor 

sensor:
  - platform: my_custom_sensor # Using the custom component
    name: "My Derived Uptime Sensor"
```

### Anything in the logs that might be useful for us?

```txt

```

echavet, 2025-06-05:
Here is the compile error when building external component MitsubishiCN105ESPHome from (https://github.com/echavet/MitsubishiCN105ESPHome) without the explicit use of an uptime component in the yaml file.

```yaml
sensor:
  # Uptime sensor added to prevent from esphome/components/uptime/sensor/uptime_seconds_sensor.h: No such file or directory
  - platform: uptime
    name: Uptime
```

```stacktrace
Compiling .pioenvs/heatpump--mayas-room/src/esphome/components/cn105/cn105.cpp.o
In file included from src/esphome.h:43,
                 from src/esphome/components/cn105/Globals.h:2,
                 from src/esphome/components/cn105/cn105.h:2,
                 from src/esphome/components/cn105/climateControls.cpp:1:
src/esphome/components/cn105/uptime_connection_sensor.h:2:10: fatal error: esphome/components/uptime/sensor/uptime_seconds_sensor.h: No such file or directory
    2 | #include <esphome/components/uptime/sensor/uptime_seconds_sensor.h>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
*** [.pioenvs/heatpump--mayas-room/src/esphome/components/cn105/climateControls.cpp.o] Error 1
In file included from src/esphome.h:43,
                 from src/esphome/components/cn105/Globals.h:2,
                 from src/esphome/components/cn105/cn105.h:2,
                 from src/esphome/components/cn105/cn105.cpp:2:
src/esphome/components/cn105/uptime_connection_sensor.h:2:10: fatal error: esphome/components/uptime/sensor/uptime_seconds_sensor.h: No such file or directory
    2 | #include <esphome/components/uptime/sensor/uptime_seconds_sensor.h>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
*** [.pioenvs/heatpump--mayas-room/src/esphome/components/cn105/cn105.cpp.o] Error 1
========================== [FAILED] Took 2.77 seconds ========================== 
```

ssieb, 2025-06-05:
Why would any component be inheriting from that?  This isn't an esphome issue.  You should not inherit from a non-platform component.

closed 2025-06-05 by ssieb

echavet, 2025-06-13:
> Why would any component be inheriting from that? This isn't an esphome issue. You should not inherit from a non-platform component.

I don't think this is the question. Even if I could try to give you an answer... 
The question could be, "why isn't the component included whereas it is specified as a custom component DEPENDENCIES" ?
Don't pay attention to the component path. This was just an example (with an error)!

ssieb, 2025-06-13:
Auto load only includes the top-level directory, not any sub-directories.