esphome/issues #7100: ili9341 white screen problem

off112a, 2025-06-05:
### The problem

Hi,

I just bought a ili9341 and got the problem with white screen. I tried with color_palette: 8BIT. The error in my log disappears but I have still the white screen. 

I'm using a esp-wrover with 8MB psram.

If I enable psram it will crash at boot and after a whole boot into safe mode. 

### Which version of ESPHome has the issue?

2025.5.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2025.5.3

### What platform are you using?

ESP32

### Board

esp32-wrover

### Component causing the issue

ili9341

### YAML Config

```yaml
esp32:
  board: esp-wrover-kit
  framework:
    type: esp-idf
    

#psram:

spi:
  clk_pin: GPIO16
  mosi_pin: GPIO17
  miso_pin: GPIO26
  id: tft_spi

display:
  - platform: ili9xxx
    model: ILI9341
    cs_pin: GPIO19
    dc_pin: GPIO32
    reset_pin: GPIO18
    data_rate: 20MHz
    rotation: 270
    invert_colors: false
    color_palette: 8BIT
    color_order: bgr
    show_test_card: true

output:
  - platform: ledc
    pin: GPIO04
    id: backlight_pwm

light:
  - platform: monochromatic
    output: backlight_pwm
    name: "Display Backlight"
    id: back_light
    restore_mode: ALWAYS_ON
```

### Anything in the logs that might be useful for us?

```txt

```

off112a, 2025-06-07:
Hi, After tested some more I finally got it to work.

When I changed the data_rate to down to 10MHz from 20MHz the display worked. Now can I also use psram without the esp crashes. 

I'm not sure what the data_rate has for function but it seems that the display works fine.

I have also rearranged the pins  but I don't think it had any impact on the result.

ssieb, 2025-06-07:
Try removing the miso pin config.

off112a, 2025-06-07:
That did the trick. I can now have data_rate for ex 40MHz. 

But I discovered that I have to have miso if I want to activate the touch.

off112a, 2025-06-07:
I have one these

[image]

ssieb, 2025-06-07:
That looks like it has separate pins for the touch.  You might be able to share the pins, but if you have the pins available, I recommend separating them.

closed 2025-06-17 by clydebarrow

off112a, 2025-06-17:
I managed to get it working with your help and some trial & error. Here is the wiring I ended up with that worked for my setup:

| eps32   |display     | touch      
| --------|------------|------------
|  5V     | VCC        |
|  GND    | GND        | 
|  2      | D/C        |
|  4      | RESET      |
|  5      | LED        |
|  15     | CS         |
|  18     | SCK        | T_CLK
|  19     |            | T_OUT
|  23     | SDI(MOSI)  | T_DIN
|  33     |            | T_CS