esphome/issues #7101: No WiFi when using "it.filled_rectangle" with ssd1306 after update ESPHomeBuilder from V2025.2.1

schemart, 2025-06-07:
### The problem

Hello,
I got exactly the same problem as already posted here:
https://github.com/esphome/issues/issues/2019:

WiFi connection can not be established, when "it.filled_rectangle" is used. There's no problem with "it.rectangle".

This problem came after an update with ESPHomeBuilder  > 2025.2.1 (don't know the breaking version exactly) 
Curent HA Version is 2025.5.3. But this version doesn't matter. 

### Which version of ESPHome has the issue?

> 2025.2.1

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP8266

### Board

esp12e

### Component causing the issue

ssd1306 / sh1106

### YAML Config

```yaml
...

captive_portal:
    
font:
  - file: "fonts/Roboto-Black.ttf"
    id: font_01
    size: 10

globals:
  - id: total_pulses
    type: int
    restore_value: False
    initial_value: '728227'  # mit diesem Wert startet der Gaszählerstand
    
binary_sensor:
  - platform: gpio
    id: internal_pulse_counter
    pin:
      number: GPIO14
      mode: INPUT
    name: "Live-Impuls"
    filters:
      - delayed_on: 100ms
    on_press:
      then:
        - lambda: id(total_pulses) += 1;
        - output.turn_on: led  # für die LED, zur Anzeige des Schaltvorgangs
    on_release:
      then:
        - output.turn_off: led  # für die LED, zur Anzeige des Schaltvorgangs
    #LED blinkt, wenn ein Signal vom Reedkontakt erkannt wird
output:
  - platform: gpio # Die interne LED auf dem Board
    pin: GPIO16
    id: 'led'
    inverted: False

sensor: 
  - platform: template
    name: "hwr-gasverbrauch"
    device_class: gas
    unit_of_measurement: "m³"
    state_class: "total_increasing"
    icon: "mdi:fire"
    accuracy_decimals: 2
    ##(0.1qm/pulse)
    #lambda: |-
    #  return id(total_pulses) * 0.01; 
      
  - platform: adc
    pin: A0
    name: "water_level_pct"
    id: water_level_pct
    unit_of_measurement: "%"
    update_interval: 900s
    #update_interval: 10s
    filters:
      #- clamp:
      #    min_value: 0.14
      #    max_value: 0.7
      #    ignore_out_of_range: true
      - multiply: 5.0
      - calibrate_linear:
        # TankDia = 198cm ->308[L/dm] x height[dm]
        # Restpegel = 0,20m
        # Überlauf über Grund = 1,14m
        # Vout = 5V*(0.09 * P + 0.04) ± 5.0% VFSS
        # P = 11,11 * (Vout/5V) - 0.04
        #~5V/m * 0,94m = 4,7V /5 (R-Teiler da ADC 1V max.)= 0.94V = max
        - 0.0 -> 0 # Wert bei 0-Druck
        #- 4.7 -> 1 #0.95m (Überlaufhöhe) with multiply 5.0
        - 4.0 -> 1 #bei Voll auf 1 bzw. 100% angepasst

i2c:
  - id: bus_a
    sda: 4
    scl: 5
    scan: false
    setup_priority: -100

display:
  - platform: ssd1306_i2c
    model: "SH1106 128x64"
    #reset_pin: D0
    i2c_id: bus_a
    address: 0x3C
    setup_priority: -100
    #rotation: 90
    flip_y: True
    update_interval: 10s
    contrast: 20%
    #DispTankwith: 40
    #DispTankheight: 60
    lambda: |-
      const int max_Vol = 2894;
      int rect_fill_height = (60 * id(water_level_pct).state);
      char buf_1[32];
      char buf_2[32];
      it.rectangle(1, 1, 40, 60);
      //it.filled_rectangle(1, (60 - rect_fill_height), 40, rect_fill_height);
      sprintf(buf_1, "Lvl: %.1f %%", (100 * id(water_level_pct).state));
      it.print(50, 15, id(font_01), buf_1);
      sprintf(buf_2, "Vol: %i Lit", int(max_Vol * id(water_level_pct).state));
      it.print(50, 30, id(font_01), buf_2);
```

### Anything in the logs that might be useful for us?

```txt

```

ssieb, 2025-06-08:
Can you get serial logs?

schemart, 2025-06-20:
Hello @ssieb,
do you already have an idea what the problem could be?

Thanks.

schemart, 2025-06-27:
Hi,
sorry for the delayed answer.
In EspHome there's no connection, even when I wait for hours.
At the moment I've deactivated the filled_rectangle and it is working without it.
I'll activate it again and try to ping it at the weekend and keep you informed.
Thx.

schemart, 2025-06-28:
Today I tried the latest version and it connects late but it connects!
But there's nothing displayed on the SH1106:
```
 INFO ESPHome 2025.6.2
INFO Reading configuration /config/esphome/esphome-web-a35b7d.yaml...
INFO Starting log output from 192.168.xx.xx using esphome API
INFO Successfully resolved iot-zisterne @ 192.168.xx.xx in 0.000s
WARNING Can't connect to ESPHome API for iot-zisterne @ 192.168.xx.xx: Error connecting to [AddrInfo(family=<AddressFamily.AF_INET: 2>, type=<SocketKind.SOCK_STREAM: 1>, proto=6, sockaddr=IPv4Sockaddr(address='192.168.xx.xx', port=6053))]: [Errno 111] Connect call failed ('192.168.xx.xx', 6053) (SocketAPIError)
INFO Trying to connect to iot-zisterne @ 192.168.xx.xx in the background
INFO Successfully resolved iot-zisterne @ 192.168.xx.xx in 0.000s
INFO Successfully resolved iot-zisterne @ 192.168.xx.xx in 0.000s
INFO Successfully resolved iot-zisterne @ 192.168.xx.xx in 0.000s
INFO Successfully resolved iot-zisterne @ 192.168.xx.xx in 0.000s
INFO Successfully resolved iot-zisterne @ 192.168.xx.xx in 0.000s
INFO Successfully resolved iot-zisterne @ 192.168.xx.xx in 0.000s
INFO Successfully connected to iot-zisterne @ 192.168.xx.xx in 1.082s
INFO Successful handshake with iot-zisterne @ 192.168.xx.xx in 3.219s
[11:06:09][I][app:137]: ESPHome version 2025.6.2 compiled on Jun 28 2025, 10:50:25
[11:06:09][C][wifi:613]: WiFi:
[11:06:09][C][wifi:434]:   Local MAC: xx:xx:xx:xx:xx:xx
[11:06:09][C][wifi:439]:   SSID: [redacted]
[11:06:09][C][wifi:442]:   IP Address: 192.168.xx.xx
[11:06:09][C][wifi:446]:   BSSID: [redacted]
[11:06:09][C][wifi:446]:   Hostname: 'iot-zisterne'
[11:06:09][C][wifi:446]:   Signal strength: -62 dB ▂▄▆█
[11:06:09][C][wifi:455]:   Channel: 6
[11:06:09][C][wifi:455]:   Subnet: 255.255.0.0
[11:06:09][C][wifi:455]:   Gateway: 192.168.xx.xx
[11:06:09][C][wifi:455]:   DNS1: 192.168.xx.xx
[11:06:09][C][wifi:455]:   DNS2: 0.0.0.0
[11:06:09][C][logger:211]: Logger:
[11:06:09][C][logger:211]:   Max Level: DEBUG
[11:06:09][C][logger:211]:   Initial Level: DEBUG
[11:06:09][C][logger:217]:   Log Baud Rate: 115200
[11:06:09][C][logger:217]:   Hardware UART: UART0
[11:06:09][C][gpio.output:010]: Binary Output:
[11:06:09][C][gpio.output:011]:   Pin: GPIO16
[11:06:10][C][template.sensor:022]: Template Sensor 'hwr-gasverbrauch'
[11:06:10][C][template.sensor:022]:   State Class: 'total_increasing'
[11:06:10][C][template.sensor:022]:   Unit of Measurement: 'm³'
[11:06:10][C][template.sensor:022]:   Accuracy Decimals: 2
[11:06:10][C][template.sensor:022]:   Device Class: 'gas'
[11:06:10][C][template.sensor:022]:   Icon: 'mdi:fire'
[11:06:10][C][template.sensor:023]:   Update Interval: 60.0s
[11:06:10][C][gpio.binary_sensor:015]: GPIO Binary Sensor 'Live-Impuls'
[11:06:10][C][gpio.binary_sensor:016]:   Pin: GPIO14
[11:06:10][C][adc.esp8266:027]: ADC Sensor 'water_level_pct'
[11:06:10][C][adc.esp8266:027]:   State Class: 'measurement'
[11:06:10][C][adc.esp8266:027]:   Unit of Measurement: '%'
[11:06:10][C][adc.esp8266:027]:   Accuracy Decimals: 2
[11:06:10][C][adc.esp8266:027]:   Device Class: 'voltage'
[11:06:10][C][adc.esp8266:031]:   Pin: GPIO17
[11:06:10][C][adc.esp8266:033]:   Samples: 1
[11:06:10][C][adc.esp8266:033]:   Sampling mode: average
[11:06:10][C][adc.esp8266:037]:   Update Interval: 900.0s
[11:06:10][C][captive_portal:089]: Captive Portal:
[11:06:10][C][esphome.ota:073]: Over-The-Air updates:
[11:06:10][C][esphome.ota:073]:   Address: iot-zisterne.local:8266
[11:06:10][C][esphome.ota:073]:   Version: 2
[11:06:10][C][esphome.ota:080]:   Password configured
[11:06:11][C][safe_mode:018]: Safe Mode:
[11:06:11][C][safe_mode:019]:   Boot considered successful after 60 seconds
[11:06:11][C][safe_mode:019]:   Invoke after 10 boot attempts
[11:06:11][C][safe_mode:019]:   Remain for 300 seconds
[11:06:11][C][api:182]: API Server:
[11:06:11][C][api:182]:   Address: iot-zisterne.local:6053
[11:06:11][C][api:187]:   Using noise encryption: YES
[11:06:11][C][mdns:122]: mDNS:
[11:06:11][C][mdns:122]:   Hostname: iot-zisterne
[11:06:11][C][i2c.arduino:072]: I2C Bus:
[11:06:11][C][i2c.arduino:073]:   SDA Pin: GPIO4
[11:06:11][C][i2c.arduino:073]:   SCL Pin: GPIO5
[11:06:11][C][i2c.arduino:073]:   Frequency: 50000 Hz
[11:06:11][C][i2c.arduino:089]:   Recovery: bus successfully recovered
[11:06:11][I][i2c.arduino:099]: Results from bus scan:
[11:06:11][I][i2c.arduino:101]: Found no devices
[11:06:11][C][ssd1306_i2c:023]: I2C SSD1306
[11:06:11][C][ssd1306_i2c:023]:   Rotations: 0 °
[11:06:11][C][ssd1306_i2c:023]:   Dimensions: 128px x 64px
[11:06:11][C][ssd1306_i2c:024]:   Address: 0x3C
[11:06:11][C][ssd1306_i2c:025]:   Model: SH1106 128x64
[11:06:11][C][ssd1306_i2c:027]:   External VCC: NO
[11:06:11][C][ssd1306_i2c:027]:   Flip X: YES
[11:06:11][C][ssd1306_i2c:027]:   Flip Y: YES
[11:06:11][C][ssd1306_i2c:027]:   Offset X: 0
[11:06:11][C][ssd1306_i2c:027]:   Offset Y: 0
[11:06:11][C][ssd1306_i2c:027]:   Inverted Color: NO
[11:06:11][D][api.connection:1570]: Home Assistant 2025.6.3 (192.168.69.31) connected
[11:06:58][I][safe_mode:042]: Boot seems successful; resetting boot loop counter
[11:07:05][D][sensor:098]: 'hwr-gasverbrauch': Sending state 7300.45020 m³ with 2 decimals of accuracy
```

Here's what I tested already (if I remember correctly):
Version			wiring sh1106		IP-Connection	           Running	Display
<= 2025.2.1		connected		OK				   OK		  OK	
>~ 2025.3		connected		nOK				   nOK		 nOK	
>~ 2025.3		disconnected	        OK				   OK		   -	
>~ 2025.6		connected	delayed but OK		   OK		nothing displayed	

The functing <it.filled_rectangle> was active in these tests.

ssieb, 2025-06-28:
You need serial logs.

schemart, 2025-06-29:
This is with <it.filled_rectangle> disabled:
```
11:38:31.819 -> rl��r��#�n������p�<����8��ǒ��p�nn��;�nĒ���b�$�rrp�n�������l�����b�n��n�䎐��b�ľ~�n�����l`���#�n��rnr���;��?�l�;p�n��܀�r���b���8�b�n��n���b�ľ~�n���r��#�n��rnr���;��?��8rr�ےn���b��`Ğ����r��nĒ`�����8�n��r���n�b��rb�������l�<��n���l`ľr�8G�l #��[0;32m[I][logger:034]: Log initialized[0m
11:38:31.944 -> [0;35m[C][safe_mode:080]: There have been 0 suspected unsuccessful boot attempts[0m
11:38:31.944 -> [0;32m[I][app:048]: Running through setup()[0m
11:38:31.944 -> [0;35m[C][adc.esp8266:020]: Running setup for 'water_level_pct'[0m
11:38:31.944 -> [0;35m[C][wifi:048]: Running setup[0m
11:38:31.944 -> [0;35m[C][wifi:061]: Starting
11:38:31.944 ->   Local MAC: xx:xx:xx:xx:xx:xx[0m
11:38:31.944 -> [0;36m[D][wifi:495]: Starting scan[0m
11:38:31.944 -> [0;33m[W][component:188]: Component wifi set Warning flag: scanning for networks[0m
11:38:32.029 -> [0;36m[D][binary_sensor:034]: 'Live-Impuls': Sending initial state ON[0m
11:38:37.949 -> [0;36m[D][wifi:510]: Found networks:[0m
11:38:37.949 -> [0;32m[I][wifi:553]: - 'FB-xx' [5m(xx:xx:xx:xx:xx:xx) [6m[0;32m▂▄▆█[0m[0m
11:38:37.949 -> [0;36m[D][wifi:555]:     Channel: 6[0m
11:38:37.949 -> [0;36m[D][wifi:556]:     RSSI: -49 dB[0m
11:38:37.949 -> [0;32m[I][wifi:553]: - 'FB-xx' [5m(xx:xx:xx:xx:xx:xx) [6m[0;33m▂▄[0;37m▆█[0m[0m
11:38:37.949 -> [0;36m[D][wifi:555]:     Channel: 6[0m
11:38:37.949 -> [0;36m[D][wifi:556]:     RSSI: -72 dB[0m
11:38:37.949 -> [0;32m[I][wifi:553]: - 'FB-xx' [5m(xx:xx:xx:xx:xx:xx) [6m[0;33m▂▄[0;37m▆█[0m[0m
11:38:37.949 -> [0;36m[D][wifi:555]:     Channel: 6[0m
11:38:37.949 -> [0;36m[D][wifi:556]:     RSSI: -73 dB[0m
11:38:37.949 -> [0;36m[D][wifi:558]: - [5m'NETGEAR'[6m [5m(xx:xx:xx:xx:xx:xx) [6m[0;33m▂▄▆[0;37m█[0m[0m
11:38:37.949 -> [0;32m[I][wifi:319]: Connecting to 'FB-xx'[0m
11:38:39.305 -> [0;32m[I][wifi:630]: Connected[0m
11:38:39.305 -> [0;35m[C][wifi:434]:   Local MAC: xx:xx:xx:xx:xx:xx[0m
11:38:39.305 -> [0;35m[C][wifi:439]:   SSID: [5m'FB-xx'[6m[0m
11:38:39.305 -> [0;35m[C][wifi:442]:   IP Address: 192.168.xx.xx[0m
11:38:39.305 -> [0;35m[C][wifi:446]:   BSSID: [5mB0:xx:xx:xx:xx:xx:xx[6m
11:38:39.305 ->   Hostname: 'iot-zisterne-test'
11:38:39.305 ->   Signal strength: -50 dB [0;32m▂▄▆█[0m[0m
11:38:39.305 -> [0;35m[C][wifi:455]:   Channel: 6
11:38:39.305 ->   Subnet: 255.255.0.0
11:38:39.305 ->   DNS2: 0.0.0.0[0m
11:38:39.305 -> [0;36m[D][wifi:639]: Disabling AP[0m
11:38:39.305 -> [0;35m[C][api:034]: Running setup[0m
11:38:39.305 -> [0;35m[C][i2c.arduino:016]: Running setup[0m
11:38:39.305 -> [0;32m[I][i2c.arduino:221]: Performing bus recovery[0m
11:38:39.305 -> [0;35m[C][ssd1306_i2c:010]: Running setup[0m
11:38:39.837 -> [0;32m[I][app:085]: setup() finished successfully![0m
11:38:39.837 -> [0;33m[W][component:203]: Component wifi cleared Warning flag[0m
11:38:39.837 -> [0;33m[W][component:188]: Component api set Warning flag: unspecified[0m
11:38:39.837 -> [0;32m[I][app:137]: ESPHome version 2025.6.2 compiled on Jun 28 2025, 16:49:55[0m
11:38:39.837 -> [0;35m[C][wifi:613]: WiFi:[0m
11:38:39.837 -> [0;35m[C][wifi:434]:   Local MAC: xx:xx:xx:xx:xx:xx[0m
11:38:39.837 -> [0;35m[C][wifi:439]:   SSID: [5m'FB-xx'[6m[0m
11:38:39.837 -> [0;35m[C][wifi:442]:   IP Address: 192.168.xx.xx[0m
11:38:39.837 -> [0;35m[C][wifi:446]:   BSSID: [5mB0:xx:xx:xx:xx:xx:xx[6m
11:38:39.837 ->   Hostname: 'iot-zisterne-test'
11:38:39.837 ->   Signal strength: -49 dB [0;32m▂▄▆█[0m[0m
11:38:39.837 -> [0;35m[C][wifi:455]:   Channel: 6
11:38:39.837 ->   Subnet: 255.255.0.0
11:38:39.837 ->   Gateway: 192.168.xx.xx
11:38:39.837 ->   DNS1: 192.168.xx.xx
11:38:39.837 ->   DNS2: 0.0.0.0[0m
11:38:39.837 -> [0;35m[C][logger:211]: Logger:
11:38:39.837 ->   Max Level: DEBUG
11:38:39.837 ->   Initial Level: DEBUG[0m
11:38:39.837 -> [0;35m[C][logger:217]:   Log Baud Rate: 115200
11:38:39.837 ->   Hardware UART: UART0[0m
11:38:39.837 -> [0;35m[C][gpio.output:010]: Binary Output:[0m
11:38:39.837 -> [0;35m[C][gpio.output:011]:   Pin: GPIO16[0m
11:38:39.837 -> [0;35m[C][template.sensor:022]: Template Sensor 'hwr-gasverbrauch'
11:38:39.837 ->   State Class: 'total_increasing'
11:38:39.837 ->   Unit of Measurement: 'm³'
11:38:39.837 ->   Accuracy Decimals: 2[0m
11:38:39.837 -> [0;35m[C][template.sensor:022]:   Device Class: 'gas'[0m
11:38:39.837 -> [0;35m[C][template.sensor:022]:   Icon: 'mdi:fire'[0m
11:38:39.837 -> [0;35m[C][template.sensor:023]:   Update Interval: 60.0s[0m
11:38:39.837 -> [0;35m[C][gpio.binary_sensor:015]: GPIO Binary Sensor 'Live-Impuls'[0m
[... 22 lines cut ...]
11:38:39.927 -> [0;35m[C][mdns:122]: mDNS:
11:38:39.927 ->   Hostname: iot-zisterne-test[0m
11:38:39.927 -> [0;35m[C][i2c.arduino:072]: I2C Bus:[0m
11:38:39.927 -> [0;35m[C][i2c.arduino:073]:   SDA Pin: GPIO4
11:38:39.927 ->   SCL Pin: GPIO5
11:38:39.927 ->   Frequency: 50000 Hz[0m
11:38:39.927 -> [0;35m[C][i2c.arduino:089]:   Recovery: bus successfully recovered[0m
11:38:39.927 -> [0;35m[C][ssd1306_i2c:023]: I2C SSD1306
11:38:39.927 ->   Rotations: 0 °
11:38:39.927 ->   Dimensions: 128px x 64px[0m
11:38:39.927 -> [0;35m[C][ssd1306_i2c:024]:   Address: 0x3C[0m
11:38:39.927 -> [0;35m[C][ssd1306_i2c:025]:   Model: SH1106 128x64[0m
[... cut ...]
11:38:39.927 ->   Inverted Color: NO[0m
11:38:39.927 -> [0;35m[C][ssd1306_i2c:036]:   Update Interval: 10.0s[0m
11:38:44.262 -> [0;33m[W][component:278]: Component display took a long time for an operation (316 ms).[0m
11:38:44.262 -> [0;33m[W][component:279]: Components should block for at most 30 ms.[0m
11:39:31.936 -> [0;32m[I][safe_mode:042]: Boot seems successful; resetting boot loop counter[0m
11:42:17.178 -> [0;36m[D][sensor:098]: 'water_level_pct': Sending state 0.00000 % with 2 decimals of accuracy[0m

```

schemart, 2025-07-23:
Hi @ssieb,
today I tried with 2025.7.3 but unfortunately no change.
If you need more informations, please let me know.

Thank you!

schemart, 2025-08-27:
Fixed since ESPHome 2025.8.1