esphome/issues #7102: Yaml code not visible anymore

maba64, 2025-06-08:
### The problem

The devices ( yaml code) have not all disappeared in the esp home. They all still work and in the homeassistant gui they are all present and the measured values are displayed. But I can no longer see or edit the yaml code. I use the latest version 

### Which version of ESPHome has the issue?

2025.5.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

Latest

### What platform are you using?

ESP32

### Board

Nodemcu

### YAML Config

```yaml

```

### Anything in the logs that might be useful for us?

```txt

```

maba64, 2025-06-08:
[image]

ssieb, 2025-06-08:
Check the addon log and check your `secrets.yaml` file.

maba64, 2025-06-09:
Your Wi-Fi SSID and password
wifi_ssid: "chrottzug"
wifi_password: "xxxxxxxxxx"
api_key: "tixpGUzptKwSMINnSmEZ+NUaThA17WCbFyvc5HeGcv4="
esp_cam_ota_password: "69f325ef1c06fed2fa6debea6670d912"
esp_cam_ap_password: "tixpGUzptKwSMINnSmEZ+NUaThA17WCbFyvc5HeGcv4="
api_key2: "tiXpGUzptKwSMINnSmEZ+NUaThA17WCbFyvc5HeGcv4="

ssieb, 2025-06-09:
Is that first line actually in the file?

maba64, 2025-06-09:
What shoold be inside this file?

ssieb, 2025-06-09:
I didn't mean delete the entire file, just that first line.