esphome/issues #7118: LVGL button: Cannot override a theme with a style

numo68, 2025-06-15:
### The problem

I have a following code. Unfortunately the whole code is quite a complicated one so cutting it down it a bit problematic, let me know if the minimal example is really needed.

I have the following:

```yaml
lvgl:
  theme:
    button:
      bg_color: 0x2F8CD8
  style_definitions:
    - id: ok_button
      bg_color: 0x00ff00

...

      widgets:
        - button:
            styles: ok_button
```

The button is blue. If I remove the `bg_color` from the theme, the button is green, which was the intent.

The generated code is
```
  lv_theme_apply_button = [](lv_obj_t * obj) -> void {
    lv_obj_set_style_bg_color(obj, lv_color_make(47, 140, 216), LV_PART_MAIN);
...
  lv_theme_apply_button(lv_btn_t_id_4);
  lv_obj_add_style(lv_btn_t_id_4, ok_button, LV_PART_MAIN);
```

Now I don't know anything but the bare minimum about LVGL, but I am afraid that the `lv_obj_set_style_bg_color` from the lambda trumps the `lv_obj_add_style`, as the former is a setting performed on a particular object, which has a larger precedence than adding a style.

The LVGL docs says "The theme for your application is a compile time configuration setting in `lv_conf.h`". I guess that either this will need to be performed in the ESPHome instead of creating and calling the lambdas, or the other way round - namely setting the styles through the objects' functions - will need to be implemented. The former is probably the canonical LVGL way.

Thanks

### Which version of ESPHome has the issue?

2025.7.0-dev

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2025.5.2

### What platform are you using?

ESP32

### Board

ESP32-WROVER-E

### Component causing the issue

lvgl

### YAML Config

```yaml

```

### Anything in the logs that might be useful for us?

```txt

```

numo68, 2025-06-15:
Alright, the `lv_conf.h` only seems to select some predefined ones, but the example at https://github.com/lvgl/lvgl/blob/release/v8.4/examples/styles/lv_example_style_14.c basically does the `lv_obj_add_style` in a theme applying callback, so perhaps just using that instead of the `lv_obj_set_style_*` might work. I am unfortunately completely lost in the lvgl python magic so I am not able to prototype it right away :(

clydebarrow, 2025-06-17:
Yes, the current implementation of themes has some shortcomings. I had done some work to make themes applied as styles instead of object properties but that may have only been in the (unreleased) LVGL 9 version. I'll look again.

clydebarrow, 2025-06-17:
This had already been implemented in the LVGL 9 version of ESPHome (which is unreleased). Backported now to the current version with the PR mentioned above.

closed 2025-06-23 by jesserockz