esphome/issues #7120: Implemente `HA suggested_display_precision` to correctly display the default decimal number starting from `HA 2025.6.0`

Sleeper85, 2025-06-16:
### The problem

Since `HA 2025.6.0`, the display of the decimal number of sensors defined with a `device_class` no longer respects the number defined in `accuracy_decimals`. It has become mandatory to define the display precision of each sensor directly in HA, which is completely stupid.

This is a problem with the display of values ​​in HA and automations linked to these values ​​but has no impact on the values ​​recorded in the statistics.

To fix this problem, it is necessary to communicate to HA the `suggested_display_precision`, which I believe should correspond to `accuracy_decimals`.

[See this article : Sensor device classes now have default display precision](https://developers.home-assistant.io/blog/2025/05/26/sensor-default-display-precision/)

[See this HA issue for more information](https://github.com/home-assistant/core/issues/146873)

### Which version of ESPHome has the issue?

2025.5.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2025.6.0

### What platform are you using?

ESP32-IDF

### Board

unrelated to a board

### YAML Config

```yaml

```

### Anything in the logs that might be useful for us?

```txt

```

ans-gw, 2025-06-21:
WARNIING...................WARNING...................WARNING.....................WARNING....................WARNING.
With this Ha update all values not have decimals.In my case from Inverter and from Bms.
let's stop and think for a moment....how many users use for example Mppsolar and Jk Bms ,or other to monitor and control the battery charge?...especially the lithium ones that burn easilyThis is very dangerous, and can lead to battery damage, or the battery can catch fire, because it is overcharged!!!  It could even burn down the house....
Who will be liable for these damages?
This problem need to be fixed quickly.
Thank you.

Running Ha on Homeassistand Blue.

P.S.: before to push update that modify something so important and dangerous ,should think twice before !

Sleeper85, 2025-06-21:
@ans-gw 

Please note this is only a problem with the display of values ​​that you can correct for each sensor by setting the number of decimal places to display. The values ​​stored in the database are correct.

ans-gw, 2025-06-21:
Yes I know that.
Can you please explain how to correct the problem with a clear example?

GHswitt, 2025-06-24:
> But watch out, even after you set the display decimal point resolution in the UI if you have to bounce your esphome device sending your sensor data or something else causes loss of signal, you can get some or all of those display resolution values being reset to 0 decimal points and you have to manually change them again.

Yes, that is very annoying. Each time after updating the device, all display precision settings are reset to default.

joshuaboniface, 2025-07-02:
I believe this fix has now caused the opposite problem in 2025.7, at least for me. Using `accuracy_decimals` to clamp the precision of a device to a given number of decimals (which, to me, is precisely what that name implies) no longer works.

For instance consider an SHT45 temperature sensor. Previously, I used `accuracy_decimals: 1` to set it to only display `24.1C` or the like, to only store values to 1 decimal place, etc.. With this update HASS is now getting some value like `24.120495920394` which is just noise, and which will also increase the rate at which the sensor updates flooding the states tracker with update events. Further now I'd need to, in any templates or custom cards, manually clamp this myself with `round(1)` statements or the like, instead of the sensor component just respecting the `accuracy_decimals`.

At the very least I'm of the opinion that ESPHome should add its own internal clamping function for this usecase, to permit device creators to define what they consider the maximum accuracy of the sensor and not report arbitrary precision.

ssieb, 2025-07-02:
Where do you see that value?

joshuaboniface, 2025-07-03:
@ssieb I'm seeing the massive decimal values in custom cards (that don't currently define a display precision; apparently this is a brand new thing I guess?) as well as in templates (I have a lot of template sensor). But I think this is honestly more of a principle thing - I don't *want* my sensors reporting 12 decimal places of precision, which is why I used `accuracy_decimals` previously. Basically, if I seem vocal about it, it's just because I don't see the point in the way this fix was done. It seems backwards to me, trying to fix an upstream issue with not enough decimals but then creating another - opposite - upstream issue with too many decimals. Maybe I'm just thinking about it wrong but upgrading and then finding all my things looking like there's suddenly a massive math error was offputting, only to then find that it was apparently intentional.

Here's some screenshots of what I mean.

Me defining `accuracy_decimals: 1` because, this is a temperature, why would I need more than 1 decimal place?

[image]

But in HASS templates, I have a ridiculous number of decimals...

[image]

And when I use that in particular in a MagicMirror template...

[image]

So I now have to go in anywhere this is displayed and manually clamp it with `|round(1)`.

Now, that was a device I control, but let's take another one for which I have a bunch of custom cards: an Athom smart plug reporting the voltage/current/power now with way too many decimals to be reasonable; I can't possibly believe the chips in these things are accurate to this many decimal places, which makes this false precision.

[image]

Now, I know the obvious solution - and yes, I plan to do this - is to just fix my custom cards to respect this new display option, and to manually clamp the value in MagicMirror tmplates. But why require people to suddenly have to modify things that were working just fine to avoid this when, well, to me, that is precisely why `accuracy_decimals` existed in the first place and why I explicitly set it in ESPHome? Configuring that on the ESPHome side shows intentionality from the developer of the device; it's not a display suggestion, unless myself and possibly others have been using it completely wrong this whole time about what "accuracy" meant. And HASS already had display precision settings on top of that.

joshuaboniface, 2025-07-03:
Yep that seems to be what changed in [home-assistant/core#147849](https://github.com/home-assistant/core/pull/147849) - previously the integration would actually clamp the value based on `accuracy_decimals`, but now it doesn't - returns the full value reported by the originating sensor while treating `accuracy_decimals` as equivalent to `suggested_display_precision`. I'm no expert on the HASS codebase but that is my thinking exactly :-)

Bunky101, 2025-07-04:
This has made a lot of work to change each and every sensor, templates, etc. HA really should make it possible to change all at once, i.e. humidity sensors have two decimal places, I would really like to have the option to change all humidity entities at once to say one decimal place. 

If it is there I haven't found it and if it isn't shame on them for not having thought of this and providing it.

jesserockz, 2025-07-04:
`{{ states("sensor.entity_id") }}` returns the raw state.

If you want a state that follows the precision, you should be using `{{ states("sensor.entity_id", rounded=True) }}`

ssieb, 2025-07-24:
You can't fix it in esphome, it's an issue in HA.

Protoncek, 2025-08-08:
> You can't fix it in esphome, it's an issue in HA.

Hm... i don't completely agree with that. It's true that you can't fix it in esphome, however, it's not a HA issue, but rather esphome one, because even if i specifically tell esphome device to use only 1 or 2 decimal it still reports raw value with 12 decimals to HA...

Bunky101, 2025-08-08:
It is not an issue with esphome. Esphome must provide the entire value. HA should have the ability to change all sensors of a type, I.e. temperature, to the same precision if desired. This is easy in software and it should be added as an option.Sent from my iPhone

Protoncek, 2025-08-09:
Yeah, as you say: filter won’t help…

ssieb, 2025-10-17:
The yaml syntax is fine.  You don't have to indent lists from the enclosing key.

> Actually, that’s not true. Esphome often compiles things around lambda even if they are totally wrong…

Do you have an example?

Protoncek, 2025-10-18:
Not right now, but i'm not an experienced in lambda, also, so when i try to do something i "do my magic" and i know that yaml can be compiled, but lambda is wrong because result is empty (not working as i want).
But, i'm talking about "things" inside lambda command itself, especially if it's more complicated. 

However regarding this indentation:
```
filters:
- lambda: return round(x *10)/10; #rounded via lambda
- round: 1 #rounded via filter
```
it might indeed be wrong.  It should be:
```
filters:
  - lambda: return round(x *10)/10; #rounded via lambda
  - round: 1 #rounded via filter
```
(if it matters...)

ssieb, 2025-10-18:
The editor can't verify your C++ inside a lambda.  That's for the compiler to do.

Both of those yaml examples are valid and equivalent.

paisitw, 2025-12-14:
Do we have any fix  on this issue yet?. I am running 2025.12 and still encountered.

jayjayvm, 2026-01-12:
I hope there is an update or a method that can be used now, for I did disable a few pages of my lovelace for it contained too many non fixable values.  even though I followed all options in HA like Display precision settings under the sensor options, round=true and round with a decimal number, none of those seem to work, never had this problem in previous much earlier versions of HA.

jayjayvm, 2026-01-13:
Tried throws an error and no value, can not read the error nor copy, else I would present it here. Thanks for thinking with me!