esphome/issues #7126: Persistent I2S Audio Data Acquisition Failure with `i2s_audio` Component on ESP32 (Arduino Framework)

vgisbert, 2025-06-18:
### The problem

I am consistently encountering issues with the `i2s_audio` component in ESPHome, which fails to acquire meaningful audio data from I2S microphones (specifically INMP441 and SPH0645) on ESP32 development boards using the Arduino framework. Despite extensive troubleshooting and testing with minimal configurations, the I2S input continuously reports invalid or no audio data, often showing `NaN` for RMS values in logs. This suggests a fundamental problem with the `i2s_audio` component's ability to initialize or process data from I2S microphones.

### Which version of ESPHome has the issue?

* **ESPHome Version: 2025.5.2 * **ESP32 Board:** ESP32 Dev Module (tested with standard ESP32 development boards) * **ESP-IDF Framework Version:** Arduino (tested with `esp32 by Espressif Systems` core versions 2.0.17 and development builds like 3.2.0) * **Microphones Tested:** INMP441 and SPH0645

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

HAOS 15.2 CORE 2025.6.1

### What platform are you using?

ESP32

### Board

ESP32_DEVKITC_V4

### Component causing the issue

i2s_audio

### YAML Config

```yaml
substitutions:
  name: "micro1_i2s_test"
  friendly_name: "micro1 I2S Test"

esphome:
  name: "${name}"
  friendly_name: "${friendly_name}"
  name_add_mac_suffix: false

esp32:
  board: esp32dev # Ensure this matches your specific ESP32 Dev Module
  framework:
    type: arduino

logger:
  level: VERBOSE # VERY IMPORTANT: VERBOSE level to see ALL details!

# -----------------------------------------------------------
# I2S AUDIO COMPONENT DEFINITION - PROBLEM AREA
# -----------------------------------------------------------
i2s_audio:
  - id: i2s_in
    i2s_lrclk_pin: GPIO27  # WS (Word Select) - LRCL on INMP441 / SPH0645
    i2s_bclk_pin: GPIO26   # SCK (Bit Clock) - BCLK on INMP441 / SPH0645

# -----------------------------------------------------------
# Microphone Configuration (uses i2s_audio as source)
# -----------------------------------------------------------
microphone:
  - platform: i2s_audio
    id: inmp441_microphone_source
    adc_type: external
    pdm: false # <<< IMPORTANT! Set to FALSE for INMP441/SPH0645 (It's an I2S microphone, not PDM directly handled by ADC)
    i2s_audio_id: i2s_in
    i2s_din_pin: GPIO13
    channel: left
    sample_rate: 16000 # Common sample rate for INMP441 (can try 8000, 44100 as well)

    # Optional: Add on_data lambda for raw sample inspection
    on_data:
      lambda: |-
        // Example: Log first few samples for debugging
        ESP_LOGD("MIC_CALLBACK", "on_data lambda del micrófono ejecutada!");
        static unsigned long last_publish_time = 0;
        const unsigned long publish_interval_ms = 50;
        if (millis() - last_publish_time >= publish_interval_ms) {
            last_publish_time = millis();
            const int16_t* samples_data = reinterpret_cast<const int16_t*>(x.data());
            size_t num_samples = x.size() / sizeof(int16_t);
            ESP_LOGD("MIC_RAW", "Tamaño del buffer de audio (bytes): %u", x.size());
            ESP_LOGD("MIC_RAW", "Número de muestras esperadas: %u", num_samples);
            if (num_samples > 0) {
                for (size_t i = 0; i < std::min((size_t)10, num_samples); ++i) {
                    ESP_LOGD("MIC_RAW", "Muestra %u: %d", i, samples_data[i]);
                }
            } else {
                ESP_LOGW("MIC_RAW", "No se recibieron muestras de audio (num_samples es 0).");
            }
        }
```

### Anything in the logs that might be useful for us?

```txt
[D][DB_CALC:142]: Raw RMS recibido: nan, Offset DB: 0.000000
[D][DB_CALC:145]: Raw RMS es 0, negativo o NaN, retornando 0.0 dB
[V][sensor:043]: 'Nivel de Ruido Calibrado dB': Received new state 0.000000
[D][sensor:094]: 'Nivel de Ruido Calibrado dB': Sending state 0.00000 dB with 1 decimals of accuracy
```

### Additional information

1.  **`Raw RMS recibido: nan` in logs:** The most consistent symptom observed in the device logs is the repeated reporting of `Raw RMS recibido: nan` (or similar debug messages indicating NaN values for raw audio input). This occurs despite the microphones being correctly wired according to their datasheets and common ESP32 I2S configurations.

vgisbert, 2025-06-21:
I've found a clue that may point to a solution to the problem.
Testing in the Arduino environment, I've found that the I2S component doesn't work with the following combination:
- Arduino IDE 2.3.6
- Expressif System ESP32 board (version 3.2.0).

The I2S component works correctly with the following combination:
- Arduino IDE 1.8.18
- Expressif System ESP32 board (version 3.2.0).

almoney, 2025-07-14:
I am still seeing the error with ESPHome 2025.6.3 and HA 2025.7.1

[i2s_audio.microphone:489]: Driver failed to start; retrying in 1 second
[06:24:48][E][component:195]: Component i2s_audio.microphone set Error flag: unspecified
[06:24:48][D][i2s_audio.speaker:136]: Starting