esphome/issues #7127: 2025.6-beta3: Can not query BLE signal strength

top-gun, 2025-06-18:
### The problem

My ESP32 uses the "esphome-tesla-ble" project. With 2025.6.0-b2 and b3 I don't get a reading for the BLE signal strength, the entity shows "unavailable". With 2025.5.2 and oder add-on versions it works just fine.

In the ESP log, I see an error instead of the usual RSSI reading:
[09:35:44][W][esp32_ble:417][BTC_TASK]: Ignoring unexpected GAP event type: 26

### Which version of ESPHome has the issue?

2025.6.0b3

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2025.6.1

### What platform are you using?

ESP32-IDF

### Board

ESP32-DevkitC (generic)

### Component causing the issue

ble_client

### YAML Config

```yaml
The YAML file has several includes. I will post one after another, I hope that's a good way. I can also post the "validate"-output if that is more helpful.

Project YAML:
'''
substitutions:
  # all substitutions for this project
  espidf_version: "5.3.1"
  platformio_version: 6.9.0
  friendly_name: Tesla BLE F549C4
  device_name: tesla-ble-f549c4
  device_description: Tesla BLE
  board: esp32dev
  variant: esp32
  flash_size: 4MB
  ble_mac_address: !secret ble_mac_address
  tesla_vin: !secret tesla_vin
  charging_amps_max: "16"

api:
  encryption:
    key: !secret api_encryption_key

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

external_components:
  source: github://PedroKTFC/esphome-tesla-ble@dev

packages:
  base:  github://PedroKTFC/esphome-tesla-ble/packages/base.yml@dev
  device_base: github://PedroKTFC/esphome-tesla-ble/boards/esp32-generic.yml@dev
  client: github://PedroKTFC/esphome-tesla-ble/packages/client.yml@dev
  # listener: github://PedroKTFC/esphome-tesla-ble/packages/listener.yml # Uncomment this to scan find your VIN BLE MAC address

esphome:
[... 327 lines cut ...]
    unit_of_measurement: "%"
    mode: slider
    update_interval: 2s
    min_value: 50
    max_value: 100
    step: 1
    set_action:
      - lambda: |-
          int var_x = static_cast<int>(x);
          id(tesla_ble_vehicle_id)->sendCarServerVehicleActionMessage(SET_CHARGING_LIMIT, var_x);
    lambda: |-
      if (id(max_soc).has_state())
      {
        return id(max_soc).state;
      }
      else return (0); 
        
---
```

### Anything in the logs that might be useful for us?

```txt
[09:35:44][W][esp32_ble:417][BTC_TASK]: Ignoring unexpected GAP event type: 26
```

slartibartfast11, 2025-06-18:
I have a similar issue after the update. I have a BLE receiver of electricity meter readings from a Powerpal device. Similar error, but it also appears to prevent the code from communicating after connecting to the device any further once the error is thrown.

top-gun, 2025-06-18:
I switched to ESP-IDF 5.3.2 and removed the platform_version statement , no improvement. I do still see "unexpected GAP event type:26"

asucrews, 2025-06-18:
Same issue, changing to mac address seems to work, but this less then ideal.

bdraco, 2025-06-18:
Looks like we need

esphome/components/esp32_ble_beacon/esp32_ble_beacon.cpp:    case ESP_GAP_BLE_ADV_DATA_RAW_SET_COMPLETE_EVT: {
esphome/components/esp32_ble_beacon/esp32_ble_beacon.cpp:    case ESP_GAP_BLE_ADV_START_COMPLETE_EVT: {
esphome/components/esp32_ble_beacon/esp32_ble_beacon.cpp:    case ESP_GAP_BLE_ADV_STOP_COMPLETE_EVT: {

closed 2025-06-19 by jesserockz

asucrews, 2025-06-19:
imported [esphome PR 9138](https://github.com/esphome/esphome/pull/9138) changes via external_components and and seems to working for my usecase

```
external_components:
  - source:
      type: git
      url: https://github.com/esphome/esphome
      ref: dev
    components: [ esp32_ble, esp32_ble_beacon, esp32_ble_tracker,bluetooth_proxy ]
    refresh: 1h
```

blair287, 2025-06-19:
This also seems to have killed bluetooth beacons. my esp32 set up to broadcast a beacon isnt detected anymore and has a load of gap errors. rolling back to 2025.05.x version of esphome fixes issue.