esphome/issues #7130: mqtt fails to resolve mdns host on 2025.6

scaiper, 2025-06-18:
### The problem

After an upgrade to 2025.6.0 and reinstalling firmware on a device, mqtt component is no longer able to resolve `.local` domain of a broker.
I have not tested if some other components are having the same issue.

If i change the `homeassistant.local` domain to the IP address everything works fine, including publishing of own hostname.

### Which version of ESPHome has the issue?

2025.6.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2025.6.1

### What platform are you using?

ESP32-IDF

### Board

C3 supermini

### Component causing the issue

mqtt

### YAML Config

```yaml
esphome:
  name: test
  friendly_name: "test"
  on_boot:
    priority: -100
    then:
      - delay: 600s
      - logger.set_level: INFO
      - logger.log:
          format: "Set log level to INFO"
          level: INFO

esp32:
  board: esp32-c3-devkitm-1
  framework:
    type: esp-idf

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  enable_btm: True
  enable_rrm: True

ota:
  - platform: esphome
    password: "redacted"

logger:

mqtt:
  broker: homeassistant.local
  port: 8883
  certificate_authority: |
    -----BEGIN CERTIFICATE-----
    redacted
    -----END CERTIFICATE-----
  username: username
  password: password
  discovery_unique_id_generator: mac

debug:
  update_interval: 60s

sensor:
  - platform: wifi_signal
    name: "WiFi Signal dB"
    id: wifi_signal_db
  - platform: uptime
    type: seconds
    name: Uptime
  - platform: debug
    loop_time:
      name: "Loop time"
  - platform: internal_temperature
    name: "Internal Temperature"

# Text sensors with general information.
text_sensor:
  # Expose WiFi information as sensors.
  - platform: wifi_info
    ip_address:
      name: IP
    ssid:
      name: SSID
    bssid:
      name: BSSID
  - platform: debug
    reset_reason:
      name: "Reset Reason"
```

### Anything in the logs that might be useful for us?

```txt
[D][mqtt:209]: Resolving broker IP address

[W][mqtt:229]: Couldn't resolve IP address for 'homeassistant.local'

[D][mqtt:209]: Resolving broker IP address

[W][mqtt:229]: Couldn't resolve IP address for 'homeassistant.local'

[D][sensor:098]: 'Uptime': Sending state 186.79500 s with 0 decimals of accuracy

[D][mqtt:209]: Resolving broker IP address

[W][mqtt:229]: Couldn't resolve IP address for 'homeassistant.local'

[D][mqtt:209]: Resolving broker IP address

[W][mqtt:229]: Couldn't resolve IP address for 'homeassistant.local'

[D][sensor:098]: 'Internal Temperature': Sending state 49.50000 °C with 1 decimals of accuracy

[D][mqtt:209]: Resolving broker IP address

[W][mqtt:229]: Couldn't resolve IP address for 'homeassistant.local'
```

scaiper, 2025-06-18:
@bdraco @kbx81 
This is when it's changed.
https://github.com/esphome/esphome/pull/8946

genehand, 2025-06-18:
This was also an issue for me with the `voice_assistant` component which connects back to the HA websocket with the default of `homeassistant.local:8123`.  Thanks for finding the workaround 👍

bioxz, 2025-06-23:
This is listed as

`Add LWIP optimization options to reduce flash usage [esphome#8946](https://github.com/esphome/esphome/pull/8946) by [@bdraco](https://github.com/bdraco) (breaking-change)`

in the changelog. It's very difficult to see (as a user) that a optimization to reduce flash usage would disable mDNS by default. And it seems even worse, I cannot even find it in ESPHome anymore because it's also not discoverable via mDNS.