esphome/issues #7137: ESP32-C2 Compiling is failing to link bootloader since 2025.6.0

Maeur1, 2025-06-19:
### The problem

Trying to upload a new firmware to an ESP32-C2 device and it appears to be failing to find a `bootloader.bin` file which seems to be failing the `merge_bin` command after compiling the firmware. Hence it cannot make a valid image to be uploaded to the ESP32-C2 switch I have.

### Which version of ESPHome has the issue?

2025.6.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2025.6.1

### What platform are you using?

ESP32-IDF

### Board

ESP8684-WROOM-04C (using the esp32-c2-devkitm-1 target)

### Component causing the issue

post_build.py script

### YAML Config

```yaml
esphome:
  name: switch-spare-bathroom
  friendly_name: Spare Bathroom Light Switch

packages:
  common: !include common/esp32.yaml

esp32:
  board: esp32-c2-devkitm-1
  framework:
    type: esp-idf
    version: latest
    sdkconfig_options:
      CONFIG_XTAL_FREQ_26: y

wifi:
  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "SpareBathSwitch Fallback Hotspot"
    password: !secret fallback_password

status_led:
  id: white_led
  pin:
    number: GPIO10
    inverted: true

switch:
  - platform: ld2410
    engineering_mode:
      name: "Engineering Mode"

output:
  - platform: gpio
    pin: GPIO6
    id: relay

light:
  - platform: binary
    name: "Spare Bathroom Lights"
    output: relay
    id: lights
    restore_mode: RESTORE_DEFAULT_ON
    
binary_sensor:
  - platform: ld2410
    has_target:
      name: Spare Bathroom Presence
    has_moving_target:
      name: Spare Bathroom Moving Target
    has_still_target:
      name: Spare Bathroom Still Target
  - platform: gpio
    internal: true
    pin:
      number: GPIO2
      mode: INPUT
      inverted: true
    name: "Switch Button"
    on_press:
      then:
        - light.toggle: lights

uart:
  id: uart_1
  tx_pin: GPIO20
  rx_pin: GPIO19
  baud_rate: 256000
  parity: NONE
  stop_bits: 1

text_sensor:
  - platform: ld2410
    version:
      name: "LD2410 Firmware Version"
    mac_address:
      name: "LD2410 MAC Address"

ld2410:

sensor:
  - platform: ld2410
    light:
      name: Spare Bathroom Light Sensor
    moving_distance:
      name : Spare Bathroom Moving Distance
    still_distance:
      name: Spare Bathroom Still Distance
    moving_energy:
      name: Spare Bathroom Move Energy
    still_energy:
      name: Spare Bathroom Still Energy
    detection_distance:
      name: Spare Bathroom Detection Distance

number:
  - platform: ld2410
    timeout:
      name: LD2410 Timeout
    light_threshold:
      name: LD2410 Light Threshold
    max_move_distance_gate:
      name: LD2410 Max Move Distance Gate
    max_still_distance_gate:
      name: LD2410 Max Still Distance Gate
    g0:
      move_threshold:
        name: LD2410 g0 Move Threshold
      still_threshold:
        name: LD2410 g0 Still Threshold
    g1:
      move_threshold:
        name: LD2410 g1 Move Threshold
      still_threshold:
        name: LD2410 g1 Still Threshold
    g2:
      move_threshold:
        name: LD2410 g2 Move Threshold
      still_threshold:
        name: LD2410 g2 Still Threshold
    g3:
      move_threshold:
        name: LD2410 g3 Move Threshold
      still_threshold:
        name: LD2410 g3 Still Threshold
    g4:
      move_threshold:
        name: LD2410 g4 Move Threshold
      still_threshold:
        name: LD2410 g4 Still Threshold
    g5:
      move_threshold:
        name: LD2410 g5 Move Threshold
      still_threshold:
        name: LD2410 g5 Still Threshold
    g6:
      move_threshold:
        name: LD2410 g6 Move Threshold
      still_threshold:
        name: LD2410 g6 Still Threshold
    g7:
      move_threshold:
        name: LD2410 g7 Move Threshold
      still_threshold:
        name: LD2410 g7 Still Threshold
    g8:
      move_threshold:
        name: LD2410 g8 Move Threshold
      still_threshold:
        name: LD2410 g8 Still Threshold

button:
  - platform: ld2410
    factory_reset:
      name: LD2410 Factory Reset
    restart:
      name: LD2410 Restart
    query_params:
      name: LD2410 Query Params

select:
  - platform: ld2410
    distance_resolution:
      name: LD2410 Distance Resolution
    baud_rate:
      name: LD2410 Baud Rate
    light_function:
      name: LD2410 Light Function
```

### Anything in the logs that might be useful for us?

```txt
INFO ESPHome 2025.6.0
INFO Reading configuration /config/esphome/switch-spare-bathroom.yaml...
INFO Generating C++ source...
INFO Updating https://github.com/espressif/esp-protocols.git@mdns-v1.8.2
INFO Compiling app...
Processing switch-spare-bathroom (board: esp32-c2-devkitm-1; framework: espidf; platform: https://github.com/pioarduino/platform-espressif32/releases/download/53.03.13/platform-espressif32.zip)
--------------------------------------------------------------------------------
HARDWARE: ESP32C2 120MHz, 272KB RAM, 4MB Flash
 - framework-arduino-c2-skeleton-lib @ 5.3.0+sha.083aad99cf 
 - framework-espidf @ 3.50302.0 (5.3.2) 
 - tool-cmake @ 3.30.2 
 - tool-esptoolpy @ 4.8.6 
 - tool-mklittlefs @ 3.2.0 
 - tool-ninja @ 1.7.1 
 - tool-riscv32-esp-elf-gdb @ 14.2.0+20240403 
 - tool-xtensa-esp-elf-gdb @ 14.2.0+20240403 
 - toolchain-esp32ulp @ 2.35.0-20220830 
 - toolchain-riscv32-esp @ 13.2.0+20240530
Reading CMake configuration...
Dependency Graph
|-- ArduinoJson @ 6.18.5
RAM:   [=         ]  11.1% (used 30824 bytes from 278528 bytes)
Flash: [=====     ]  49.1% (used 900960 bytes from 1835008 bytes)
Building .pioenvs/switch-spare-bathroom/firmware.bin
Creating esp32c2 image...
Successfully created esp32c2 image.
esp32_create_combined_bin([".pioenvs/switch-spare-bathroom/firmware.bin"], [".pioenvs/switch-spare-bathroom/firmware.elf"])
usage: esptool merge_bin [-h] --output OUTPUT [--format {raw,uf2,hex}]
                         [--chunk-size CHUNK_SIZE] [--md5-disable]
                         [--flash_freq {keep,80m,60m,48m,40m,30m,26m,24m,20m,16m,15m,12m}]
                         [--flash_mode {keep,qio,qout,dio,dout}]
                         [--flash_size {keep,256KB,512KB,1MB,2MB,2MB-c1,4MB,4MB-c1,8MB,16MB,32MB,64MB,128MB}]
                         [--spi-connection SPI_CONNECTION]
                         [--target-offset TARGET_OFFSET]
                         [--fill-flash-size {256KB,512KB,1MB,2MB,4MB,8MB,16MB,32MB,64MB,128MB}]
                         <address> <filename> [<address> <filename> ...]
esptool merge_bin: error: argument <address> <filename>: [Errno 2] No such file or directory: '/data/build/switch-spare-bathroom/.pioenvs/switch-spare-bathroom/bootloader.bin'
*** [.pioenvs/switch-spare-bathroom/firmware.bin] Explicit exit, status 2
========================== [FAILED] Took 3.72 seconds ==========================
```

### Additional information

This was working perfectly fine and is still running fine on 2025.5.2 and I think it might be something to do with the esp-idf upgrade to 5.3.2? I do know this chip is currently unsupported but this was working fine in the previous version and some change to the FLASH_ARGs seems to have broken this?

Maeur1, 2025-06-30:
Just tried this on 2025.6.2 also and its failing for me also there.

Maeur1, 2025-07-03:
Even though the changelog didn't have any updates regarding this, 2025.6.3 also continues to not function, I wonder what this missing bootloader.bin file exactly is?

Maeur1, 2025-07-16:
Ok this issue is still persisting in 2025.7.0. BUT I found a solution:

Modifying the `post_build.py.script` to be the following:

```
Import("env")

import os
import json
import pathlib

firmware_name = os.path.basename(env.subst("$PROGNAME")) + ".bin"
build_dir = pathlib.Path(env.subst("$BUILD_DIR"))
firmware_path = build_dir / firmware_name
flash_size = env.BoardConfig().get("upload.flash_size", "4MB")

# Collect flash sections to merge
sections = []
flasher_args_path = build_dir / "flasher_args.json"

if flasher_args_path.exists():
    try:
        with flasher_args_path.open() as f:
            flash_data = json.load(f)
        for addr, fname in sorted(flash_data["flash_files"].items(), key=lambda kv: int(kv[0], 16)):
            path = pathlib.Path(fname)
            if path.exists():
                sections.append((addr, str(path)))
            else:
                print(f"Info: {path.name} not found — skipping")
    except Exception as e:
        print(f"Warning: Failed to parse flasher_args.json — {e}")
else:
    print("Warning: flasher_args.json not found — falling back to legacy guesses")
    guesses = [
        ("0x0", build_dir / "bootloader" / "bootloader.bin"),
        ("0x8000", build_dir / "partition_table" / "partition-table.bin"),
        ("0xe000", build_dir / "ota_data_initial.bin"),
        ("0x10000", firmware_path)
    ]
    for addr, path in guesses:
        if path.exists():
            sections.append((addr, str(path)))
        else:
            print(f"Info: {path.name} not found — skipping")

# Only run esptool if we found valid sections
if not sections:
    print("No valid flash sections found — skipping .factory.bin creation.")
else:
    output_path = firmware_path.with_suffix(".factory.bin")
    cmd = [
        "merge_bin",
        "--fill-flash-size", flash_size,
        "--output", str(output_path)
    ]
    for addr, path in sections:
        cmd += [addr, path]

    print(f"Merging binaries into {output_path}")
    result = env.Execute(
        env.VerboseAction(
            f"{env.subst('$PYTHONEXE')} -m esptool " + " ".join(cmd),
            "Merging binaries with esptool"
        )
    )

    if result == 0:
        print(f"Successfully created {output_path}")
    else:
        print(f"Error: esptool merge_bin failed with code {result}")
```
This appears to work with the C2 + all the existing chips

I wrote that file to `./post_build.py.script` and then ran `docker cp ./post_build.py.script addon_5c53de3b_esphome:/esphome/esphome/components/esp32/post_build.py.script` which injected it into the existing docker container and that made esphome fully compile and successfully upload it to the C2 micro (exact same YAML as what I have in the original post) 

I haven't gotten around to making a PR namely because I have zero close what has in the script, I just used some AI coding tools to help me fix the script so I am gonna first understand what was written then open a PR (its late tonight here in AU) Or if someone else more clever than me understands whats going on then happy for someone else to submit the PR while I sleep.

ssieb, 2025-07-16:
Can you show a diff?  What did you change?

Maeur1, 2025-07-16:
> Can you show a diff? What did you change?

The diff is quite significant since the script hadnt updated for like a year, but it appears that it handles cases when the merge_bin step needs to be performed or not. It handles it on a case by case basis (a bit too much logging but nevertheless it still allows for both types of ESP32 MCUs)

RaceNJason, 2025-08-02:
When using the 9578 PR it now just says I have not specified a valid platform for the ESP32-C2 chip. Remarking it out allows the build up until the final merge as in original post. So I'm not sure what the PR9578 is suppose to fix...but, it doesn't appear to fix this issue for me.