esphome/issues #7145: ESP Home Entity not available in Deep Sleep after update to 2025.6.0

dergib22, 2025-06-20:
### The problem

I have an esp8266 (esp_wroom_02) that deep sleeps for 1 hour after taking readings for a few minutes. When going into deep sleep, the esp controller triggers a timer chip which cuts the 3.3VDC supplied to the esp8266.

Prior to update, HA maintained the related entities while the controller was in deep sleep.
After updating the flashed ESPHome to 2025.6.0, HA shows the related entities as unavailable.

I tried removing and re-adding the device in Home Assistant, and the entities still went unavailable.
Rolling back to 2025.5.2 resulted in the desired behavior.

I do see that there change made to the deep sleep/reboot for version 2025.6.0: https://github.com/esphome/esphome/pull/9008

### Which version of ESPHome has the issue?

2025.6.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2025.6.1

### What platform are you using?

ESP8266

### Board

ESP-WROOM-02

### Component causing the issue

deep_sleep

### YAML Config

```yaml

```

### Anything in the logs that might be useful for us?

```txt

```

Elekesey, 2025-06-28:
Exactly same issue there with same specs.

DavidSichau, 2025-07-01:
Also esp32 is affected since Core 2025.6.0

dergib22, 2025-07-11:
I am working on aquiring an assortment of parts to try replicating this behavior on hardware that is not actively deployed around my house.

dergib22, 2025-07-16:
Edited to add a minimal version of code that I am seeing this issue with. Along with the very verbose log from flashing and testing this code on my controller.

I have not been able to reliably replicate this issue on a second ESP8266, but I suspect this is possibly due to using an independent ESP32-C3 to process the 'Done' signal before it turns off the ESP8266's 3.3VDC regulator.  

To me it is feeling like there is a race condition occurring between tearing down the API connection and cutting the power to the ESP8266 that did not exist in ESPHome 2025.5.2. If the power is cut to soon, I suspect the tear down doesn't complete, and then the entity eventually goes 'unavailable'.

ssieb, 2025-07-17:
Yes, if HA doesn't get the shutdown message before the device turns off, then it will mark the device unavailable.
That shutdown trigger is very high priority.  But also, if that triggers a power off right away, then you're probably going to lose the message over wifi.  Try making the priority negative and if that isn't enough, add a `delay(500)` in a lambda before turning on the switch.

dergib22, 2025-07-17:
Adding a `delay(10)` in a lambda seems to fix my issue.

```yaml
  on_shutdown: 
    - priority: 800
      then:
      - if:
          condition:
            - switch.is_on: deep_sleep_trig
          then:
            - lambda: 'delay(10);'
            - switch.turn_on: Done
  ```

I did not try any other delay duration.
I did try a `-100` priority which caused the issue to return. This is inline with the `on_shutdown` documentation which says: 
> A higher value means a high priority and in case of shutdown triggers that the code is executed later.