esphome/issues #7150: WT32 ETH01 Compilation error at 2025.6.0

tomik671108, 2025-06-21:
### The problem

I am getting error messages during compilation 2025.6.0, I suspect it is related to ESP-IDF 5.3.2
Will this bug be fixed in the next version 5.3.3 or do I have to make the changes myself as mentioned here ?.:

[](https://github.com/espressif/esp-idf/commit/e9df36a2dfe9ee074f421aedaff8952a348b339d#diff-d486024b46fb442e1174545fc43447df3cf6b4955c32bc3164f19e34fb3b1ef1)

### Which version of ESPHome has the issue?

2025.6.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2025.5.3

### What platform are you using?

ESP32-IDF

### Board

WT32 ETH01

### Component causing the issue

ethernet_component.cpp

### YAML Config

```yaml
substitutions:
  devicename: heatpump2
  description: Heatpump Controller

esphome:
  name: "${devicename}"
  comment: "${description}"
  friendly_name: "${description}"
# Platformio options konieczne przy framework type.: esp-idf aby działał LAN na WT32ETH01
  platformio_options:
    board_build.flash_mode: dio
    board_build.flash_size: keep
  project:
    name: "${devicename}.${description}"
    version: 7.1.2

esp32:
  board: wt32-eth01
  framework:
    type: esp-idf

# Enable logging
logger:
  level: INFO
  baud_rate: 0

# Ethernet config
ethernet:
  type: LAN8720
  mdc_pin: GPIO23
  mdio_pin: GPIO18
  clk_mode: GPIO0_IN
  phy_addr: 1
  power_pin: GPIO16
  use_address: 192.168.201.10

  manual_ip:
    static_ip: 192.168.201.10
    gateway: 192.168.201.1
    subnet: 255.255.255.0
#    dns1: 192.168.200.1
```

### Anything in the logs that might be useful for us?

```txt
src/esphome/components/ethernet/ethernet_component.cpp: In member function 'virtual void esphome::ethernet::EthernetComponent::setup()':
src/esphome/components/ethernet/ethernet_component.cpp:134:21: warning: 'eth_esp32_emac_config_t::<unnamed union>::<unnamed struct>::smi_mdc_gpio_num' is deprecated: Please use smi_gpio instead [-Wdeprecated-declarations]
  134 |   esp32_emac_config.smi_mdc_gpio_num = this->mdc_pin_;
      |                     ^~~~~~~~~~~~~~~~
In file included from /data/cache/platformio/packages/framework-espidf/components/esp_eth/include/esp_eth_driver.h:13,
                 from /data/cache/platformio/packages/framework-espidf/components/esp_eth/include/esp_eth.h:15,
                 from src/esphome/components/ethernet/ethernet_component.h:10,
                 from src/esphome/components/ethernet/ethernet_component.cpp:1:
/data/cache/platformio/packages/framework-espidf/components/esp_eth/include/esp_eth_mac_esp.h:171:17: note: declared here
  171 |             int smi_mdc_gpio_num __attribute__((deprecated("Please use smi_gpio instead")));  /*!< SMI MDC GPIO number, set to -1 could bypass the SMI GPIO configuration */
      |                 ^~~~~~~~~~~~~~~~
src/esphome/components/ethernet/ethernet_component.cpp:134:21: warning: 'eth_esp32_emac_config_t::<unnamed union>::<unnamed struct>::smi_mdc_gpio_num' is deprecated: Please use smi_gpio instead [-Wdeprecated-declarations]
  134 |   esp32_emac_config.smi_mdc_gpio_num = this->mdc_pin_;
      |                     ^~~~~~~~~~~~~~~~
/data/cache/platformio/packages/framework-espidf/components/esp_eth/include/esp_eth_mac_esp.h:171:17: note: declared here
  171 |             int smi_mdc_gpio_num __attribute__((deprecated("Please use smi_gpio instead")));  /*!< SMI MDC GPIO number, set to -1 could bypass the SMI GPIO configuration */
      |                 ^~~~~~~~~~~~~~~~
src/esphome/components/ethernet/ethernet_component.cpp:134:21: warning: 'eth_esp32_emac_config_t::<unnamed union>::<unnamed struct>::smi_mdc_gpio_num' is deprecated: Please use smi_gpio instead [-Wdeprecated-declarations]
  134 |   esp32_emac_config.smi_mdc_gpio_num = this->mdc_pin_;
      |                     ^~~~~~~~~~~~~~~~
/data/cache/platformio/packages/framework-espidf/components/esp_eth/include/esp_eth_mac_esp.h:171:17: note: declared here
  171 |             int smi_mdc_gpio_num __attribute__((deprecated("Please use smi_gpio instead")));  /*!< SMI MDC GPIO number, set to -1 could bypass the SMI GPIO configuration */
      |                 ^~~~~~~~~~~~~~~~
src/esphome/components/ethernet/ethernet_component.cpp:135:21: warning: 'eth_esp32_emac_config_t::<unnamed union>::<unnamed struct>::smi_mdio_gpio_num' is deprecated: Please use smi_gpio instead [-Wdeprecated-declarations]
  135 |   esp32_emac_config.smi_mdio_gpio_num = this->mdio_pin_;
      |                     ^~~~~~~~~~~~~~~~~
/data/cache/platformio/packages/framework-espidf/components/esp_eth/include/esp_eth_mac_esp.h:172:17: note: declared here
  172 |             int smi_mdio_gpio_num __attribute__((deprecated("Please use smi_gpio instead"))); /*!< SMI MDIO GPIO number, set to -1 could bypass the SMI GPIO configuration */
      |                 ^~~~~~~~~~~~~~~~~
src/esphome/components/ethernet/ethernet_component.cpp:135:21: warning: 'eth_esp32_emac_config_t::<unnamed union>::<unnamed struct>::smi_mdio_gpio_num' is deprecated: Please use smi_gpio instead [-Wdeprecated-declarations]
  135 |   esp32_emac_config.smi_mdio_gpio_num = this->mdio_pin_;
      |                     ^~~~~~~~~~~~~~~~~
/data/cache/platformio/packages/framework-espidf/components/esp_eth/include/esp_eth_mac_esp.h:172:17: note: declared here
  172 |             int smi_mdio_gpio_num __attribute__((deprecated("Please use smi_gpio instead"))); /*!< SMI MDIO GPIO number, set to -1 could bypass the SMI GPIO configuration */
      |                 ^~~~~~~~~~~~~~~~~
src/esphome/components/ethernet/ethernet_component.cpp:135:21: warning: 'eth_esp32_emac_config_t::<unnamed union>::<unnamed struct>::smi_mdio_gpio_num' is deprecated: Please use smi_gpio instead [-Wdeprecated-declarations]
  135 |   esp32_emac_config.smi_mdio_gpio_num = this->mdio_pin_;
      |                     ^~~~~~~~~~~~~~~~~
/data/cache/platformio/packages/framework-espidf/components/esp_eth/include/esp_eth_mac_esp.h:172:17: note: declared here
  172 |             int smi_mdio_gpio_num __attribute__((deprecated("Please use smi_gpio instead"))); /*!< SMI MDIO GPIO number, set to -1 could bypass the SMI GPIO configuration */
```

ssieb, 2025-06-21:
Those are only warnings.  Is the compile failing?

tomik671108, 2025-06-22:
Seeing such messages I stopped compilation for fear of losing communication with board.

swoboda1337, 2025-06-22:
Its just a warning. Fix is easy, I already included it in this PR: https://github.com/esphome/esphome/pull/8457/files#diff-842a35cb51d8a9c418e5c970654744861b96b719435dfa1fd3d2ae8bbab754d3R148

tomik671108, 2025-06-22:
Thanks for the clarification.
Should I change something in the configuration to eliminate these warnings?

swoboda1337, 2025-06-22:
You can just ignore them :)

closed 2025-11-19 by github-actions[bot]