esphome/issues #7151: Olimex POE Stack overflow, followed by crashes and Ethernet stack becoming unresponsive

jb123567, 2025-06-22:
### The problem

All 3 of my Olimex POE ESP32 and a WT01 ETH appear to have developed an issue where they are rebooting regularly before the ethernet crashing crashes completely requiring a power cycle to bring it back up. I have temporarily moved the devices to Wi-Fi and they have now been completely stable. This behaviour seems to have started around 2025.4

### Which version of ESPHome has the issue?

2025.6

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2025.6

### What platform are you using?

ESP32

### Board

Olimex ESP32 POE ISO

### Component causing the issue

Ethernet

### YAML Config

```yaml
esphome:
  name: ajaxcontroller
  
esp32:
  board: esp32dev

# Enable logging
logger:

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  fast_connect: True

api:

ota:
  platform: esphome
  password: 

mqtt:
  broker: 
  username: 
  password: 
  client_id: AjaxAlarmControl
  topic_prefix: ajax/alarm/control
  on_message:
    - topic: ajax/alarm
      payload: arm_zone1
      qos: 1
      then:
        - switch.turn_on: arm
    - topic: ajax/alarm
      payload: disarm_zone1
      qos: 1
      then:
        - switch.turn_on: disarm
    - topic: ajax/alarm
      payload: arm_zone1_night
      qos: 1
      then:
        - switch.turn_on: arm_night
        
switch:
  - platform: gpio
    pin: 
      number: GPIO13
      inverted: true
    id: "disarm"
    on_turn_on:
      - delay: 300ms
      - switch.turn_off: disarm
  - platform: gpio
    pin:
      number: GPIO33
      inverted: true
    id: "arm"
    on_turn_on:
      - delay: 300ms
      - switch.turn_off: arm
  - platform: gpio
    pin: 
      number: GPIO32
      inverted: true
    id: "arm_night"
    on_turn_on:
       - delay: 300ms
       - switch.turn_off: arm_night
       
sensor:
  - platform: uptime
    type: seconds
    name: Uptime Sensor
#text_sensor:
 # - platform: ethernet_info
  #  ip_address:
   #   name: IP Address
        
ethernet:
  type: LAN8720
  mdc_pin: GPIO23
  mdio_pin: GPIO18
  clk_mode: GPIO17_OUT
  phy_addr: 0
  power_pin: GPIO12
```

### Anything in the logs that might be useful for us?

```txt
`[08:13:27][D][sensor:099]: 'Uptime Sensor': Sending state 2523.43408 s with 0 decimals of accuracy
[08:13:35]Guru Meditation Error: Core  1 panic'ed (Unhandled debug exception). 
[08:13:35]Debug exception reason: Stack canary watchpoint triggered (emac_rx) 
[08:13:35]Core  1 register dump:
[08:13:35]PC      : 0x4008db13  PS      : 0x00060036  A0      : 0x8008c468  A1      : 0x3ffc7e20
[08:13:35]A2      : 0x3ffbef08  A3      : 0xb33fffff  A4      : 0x0000cdcd  A5      : 0x00060023
[08:13:35]A6      : 0x00060023  A7      : 0x0000abab  A8      : 0xb33fffff  A9      : 0xffffffff
[08:13:35]A10     : 0x00000030  A11     : 0x3ffc80c0  A12     : 0x800845fa  A13     : 0x3ffbee5c
[08:13:35]A14     : 0x007bef08  A15     : 0x003fffff  SAR     : 0x00000004  EXCCAUSE: 0x00000001
[08:13:35]EXCVADDR: 0x00000000  LBEG    : 0x40088789  LEND    : 0x40088799  LCOUNT  : 0xfffffffb
[08:13:35]
[08:13:35]
[08:13:35]Backtrace:0x4008db10:0x3ffc7e200x4008c465:0x3ffc7e60 0x4008aab0:0x3ffc7e90 0x4008aa60:0x3f4022d8  |<-CORRUPTED
[08:13:35]
[08:13:35]
[08:13:35]
[08:13:35]
[08:13:35]ELF file SHA256: 0000000000000000
[08:13:35]
[08:13:35]Rebooting...
[08:13:35]ets Jul 29 2019 12:21:46
[08:13:35]
[08:13:35]rst:0xc (SW_CPU_RESET),boot:0x1b (SPI_FAST_FLASH_BOOT)
[08:13:35]configsip: 0, SPIWP:0xee
[08:13:35]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[08:13:35]mode:DIO, clock div:2
[08:13:35]load:0x3fff0030,len:1184
[08:13:35]load:0x40078000,len:13132
[08:13:35]load:0x40080400,len:3036
[08:13:35]entry 0x400805e4
[08:13:35][I][logger:171]: Log initialized
[08:13:35][C][safe_mode:080]: There have been 0 suspected unsuccessful boot attempts�fj5
[08:13:35][D][esp32.preferences:143]: Writing 1 items: 0 cached, 1 written, 0 failed`
```

drewancil, 2025-07-11:
It looks from your config that you have the wroom module and not the wrover.  I have about 10 of these and have not experienced any issues as you describe.  Only difference I can see is I set the board type to ‘esp32-poe’ or ‘esp32-poe-iso’.   Maybe try that first?

jb123567, 2025-07-11:
> It looks from your config that you have the wroom module and not the wrover.  I have about 10 of these and have not experienced any issues as you describe.  Only difference I can see is I set the board type to ‘esp32-poe’ or ‘esp32-poe-iso’.   Maybe try that first?

I bet that is going to be it, so simple can't believe I didn't spot that!! Than you, will test and see how I get on!