esphome/issues #7154: ESP8266 Infrared binary sensors stopped triggering in ESPHome 2025.6.0

madbrain76, 2025-06-22:
### The problem

After upgrading from ESPHome 2025.5.2 to 2025.6.0, and flashing my four AtHom AR-01 wirelessly, all of them stopped working. My automations use the receiver part only. The ESPHome device log still showed them receiving something from my remote, but the plumbing for the corresponding HA automations did not work anymore. The result is that I could no longer control my smart lights using my IR remotes in four different rooms.

I had to rollback to 2025.5.2 and reflash all 4 devices to get it to work.

### Which version of ESPHome has the issue?

2025.6.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

Core 2025.6.1

### What platform are you using?

ESP8266

### Board

AtHom AR-01

### Component causing the issue

unsure

### YAML Config

```yaml
substitutions:
  name: "ht-infrared"
  friendly_name: "Home theater infrared"

esphome:
  name: "${name}"
  friendly_name: "${friendly_name}"

esp8266:
  board: esp8285
  restore_from_flash: true

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: <removed>

ota:
  - platform: esphome

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  use_address: ht-infrared.localdomain
  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "ht-infrared"
    password: <removed>

captive_portal:

web_server:
  port: 80

sensor:
  - platform: wifi_signal
    name: "${friendly_name} WiFi Signal"
    update_interval: 60s

  - platform: uptime
    name: "${friendly_name} Uptime"

text_sensor:
  - platform: wifi_info
    ip_address:
      name: "${friendly_name} IP Address"
      disabled_by_default: true
    ssid:
      name: "${friendly_name} SSID"
    bssid:
      name: "${friendly_name} BSSID"
    mac_address:
      name: "${friendly_name} Mac Wifi Address"
    scan_results:
      name: "${friendly_name} Latest Scan Results"

switch:
  - platform: restart
    name: "${friendly_name} restart"

binary_sensor:
  - platform: status
    name: "${friendly_name} IR Status"

  - platform: gpio
    pin: GPIO0
    id: physical_button

  - platform: remote_receiver
    name: "Ceiling lights off"
    lg:
      data: 0x08E76897
      nbits: 32

  - platform: remote_receiver
    name: "Ceiling lights on"
    lg:
      data: 0x08E7708F
      nbits: 32

  - platform: remote_receiver
    name: "Recessed lights off"
    lg:
      data: 0x08E750AF
      nbits: 32

  - platform: remote_receiver
    name: "Recessed lights on"
    lg:
      data: 0x08E7A857
      nbits: 32

  - platform: remote_receiver
    name: "Eyeball lights off"
    lg:
      data: 0x08E7906F
      nbits: 32

  - platform: remote_receiver
    name: "Eyeball lights on"
    lg:
      data: 0x08E728D7
      nbits: 32

  - platform: remote_receiver
    name: "All lights off"
    lg:
      data: 0x08E710EF
      nbits: 32

  - platform: remote_receiver
    name: "All lights on"
    lg:
      data: 0x08E730CF
      nbits: 32

status_led:
  pin: GPIO13

remote_transmitter:
  pin:
    number: GPIO4
  carrier_duty_percent: 50%

remote_receiver:
  pin:
    number: GPIO5
    inverted: true
  dump: all
```

### Anything in the logs that might be useful for us?

```txt
N/A
```

### Additional information

This was also impacting several others in the HA community.

See 
https://community.home-assistant.io/t/regression-in-2025-6-0-with-athom-ar-01-infrared-receiver-blaster/903859

https://community.home-assistant.io/t/remote-receiver-binary-sensors-stopped-updating/903430

Changing the binary sensors to add filters / delay_off: 100 ms works around the regression.

binary_sensor:
  - platform: status
    name: "${friendly_name} IR Status"

  - platform: gpio
    pin: GPIO0
    id: physical_button

  - platform: remote_receiver
    name: "Ceiling lights off"
    lg:
      data: 0x08E76897
      nbits: 32
    filters:
      - delayed_off: 100ms

  - platform: remote_receiver
    name: "Ceiling lights on"
    lg:
      data: 0x08E7708F
      nbits: 32
    filters:
      - delayed_off: 100ms

  - platform: remote_receiver
    name: "Recessed lights off"
    lg:
      data: 0x08E750AF
      nbits: 32
    filters:
      - delayed_off: 100ms

  - platform: remote_receiver
    name: "Recessed lights on"
    lg:
      data: 0x08E7A857
      nbits: 32
    filters:
      - delayed_off: 100ms

  - platform: remote_receiver
    name: "Eyeball lights off"
    lg:
      data: 0x08E7906F
      nbits: 32
    filters:
      - delayed_off: 100ms

  - platform: remote_receiver
    name: "Eyeball lights on"
    lg:
      data: 0x08E728D7
      nbits: 32
    filters:
      - delayed_off: 100ms

  - platform: remote_receiver
    name: "All lights off"
    lg:
      data: 0x08E710EF
      nbits: 32
    filters:
      - delayed_off: 100ms

  - platform: remote_receiver
    name: "All lights on"
    lg:
      data: 0x08E730CF
      nbits: 32
    filters:
      - delayed_off: 100ms

ssieb, 2025-06-23:
You should use events instead of trying to detect a binary sensor.

madbrain76, 2025-06-24:
> You should use events instead of trying to detect a binary sensor.

Could you please explain how to do that ?

Regardless, this is a regression impact many users, and I believe it should be fixed.

ssieb, 2025-06-24:
```yaml
  - platform: remote_receiver
    name: "Ceiling lights off"
    lg:
      data: 0x08E76897
      nbits: 32
    on_press:
     - homeassistant.event:
        event: esphome.ir_receive
        data:
          message: ceiling_lights_off
```
You can use the event as an automation trigger in HA.
Using the event component is a bit more complicated, but probably better.  If you need help with that, come ask on discord.

Maeur1, 2025-06-24:
> https://esphome.io/components/api#homeassistant-event-action or https://esphome.io/components/#event-components
> 
> I expect this has to do with the api queuing fixes. If the same component publishes too quickly, only the last message will get sent. It might be considered a regression by some people, but I doubt it would be fixed.

This probably explains why the remote receiver fails to work. It makes a binary sensor go "on" then "off" really quickly and this means HA never gets the message from ESPHome even though I can see in the logs from the ESPHome device itself. 

This means any automation in HA even if it is event based or device based doesn't see the state transition.

madbrain76, 2025-06-24:
> - platform: remote_receiver
>     name: "Ceiling lights off"
>     lg:
>       data: 0x08E76897
>       nbits: 32
>     on_press:
>      - homeassistant.event:
>         event: esphome.ir_receive [...]

Thank you.  I am using an HA blueprint for my remote. It has 20 inputs for binary sensors, one for each on and each off button. And 10 inputs for corresponding switches or light entities to turn on or off.

I then create 4 automations based on this blueprint, with the binary sensors from each ESPHome AtHome device, and lights/switches I want to control.

I could rewrite the blueprint to use event instead of device (binary sensor) triggers. But it sounds like that also won't work based on Maeur's comment immediately above mine.

mahajana13, 2025-06-24:
> - platform: remote_receiver
>     name: "Ceiling lights off"
>     lg:
>       data: 0x08E76897
>       nbits: 32
>     on_press:
>      - homeassistant.event:
>         event: esphome.ir_receive [...]

Thank you very much. That helps

ssieb, 2025-06-24:
I do recommend looking at the event component.  I believe that gives you actual targets in HA instead of having to fill them in yourself with the general event sending that I showed.

mahajana13, 2025-06-25:
i took the name from:

platform: remote_receiver
name: ###

and place it to:

on_press:
message: ###

i didn´t have to change the Automations. It works. Thanks so much

bdraco, 2025-06-25:
> [esphome.io/components/api#homeassistant-event-action](https://esphome.io/components/api#homeassistant-event-action) or [esphome.io/components#event-components](https://esphome.io/components/#event-components)
> 
> I expect this has to do with the api queuing fixes. If the same component publishes too quickly, only the last message will get sent. It might be considered a regression by some people, but I doubt it would be fixed.

This behavior was an expected outcome of the state batching, which is why [PR #9012](https://github.com/esphome/esphome/pull/9012) was marked as a breaking change. Previously, if the transmit buffer was full, earlier messages could already be dropped—so the behavior was inconsistent. Now, batching ensures consistent behavior, but only the latest state gets sent when updates happen too quickly.

That said, batching is configurable and can be adjusted or disabled if needed, depending on the use case.

You're absolutely right that using events is the correct solution here. It provides consistent results regardless of whether batching is enabled or not, and avoids the limitations of state updates.

jkosharek, 2025-06-25:
> This behavior was an expected outcome of the state batching, which is why [PR #9012](https://github.com/esphome/esphome/pull/9012) was marked as a breaking change. Previously, if the transmit buffer [...]

Are there examples of how to change the ESPHome configurations to use events and how to use those events to trigger an automation? 
I added this on_press to my IR command but can't get it to trigger the automation
```
  - platform: remote_receiver
    name: "Flush Small"
    receiver_id: remote_receiver_input
    toto:
      command: 0x88
      rc_code_1: 0x0
      rc_code_2: 0x0
    on_press:
     - homeassistant.event:
        event: esphome.ir_receive
        data:
          message: master_flush_small
```

madbrain76, 2025-06-25:
For me, it is a non-trivial amount of work to change from binary sensors to events.

Each of the 4 ESPHome device configurations has 8 binary sensors. That's 32 changes.

My HA blueprint takes up to 20 binary sensors as inputs, and acts on these 20 possible state changes. That's 20 changes.

Each of the 4 automations actually fills in 8 of these binary sensors. That's 32 changes.

So, we are talking  about 84 config changes here.

I wish there was a simple way to convert it all. But for now, I'm using the 100ms delay as a workaround.

I just backed up my config and rolled ESPHome back to 2025.5.2 , to check what it shows when upgrading. I don't see any trace of a breaking change being listed for the proposed upgrade to 2025.6.1 .

[image]

jesserockz, 2025-06-25:
> I don't see any trace of a breaking change being listed for the proposed upgrade to 2025.6.1 .

@madbrain76 Unfortunately due to the way the Home Assistant add-on changelog works, it only show the notes from the latest release and not the releases in between that you skipped/missed like 2025.6.0.

If you click `Read release announcement` though, that will take you to the full release notes where the breaking change PR is listed.

bdraco, 2025-06-27:
While events are the right choice for something like this as 

> For me, it is a non-trivial amount of work to change from binary sensors to events.

Totally understandable — that’s a non-trivial amount of work, and I can see why the 100ms delay workaround makes sense for now.

If you haven't already, you could try lowering the [`batch_delay`](https://esphome.io/components/api.html#api-batch_delay) in the API component to something like `10ms` or `20ms`. That way, you still get the benefits of batching without adding too much latency.

That said, using binary sensors to represent events does carry some risk. It depends on specific timing, which can *easily shift* due to batching, system load, the state of the transmit buffer, or changes in ESPHome’s behavior down the road. Events are more robust for capturing discrete actions, since they aren't affected by how and when state updates get sent.

So your current workaround is totally fair, but if you ever run into flakiness again, switching to events could help avoid some of those edge cases.

When working with rapid or one-shot triggers, such as button presses, motion activations, or other momentary inputs, more reliable to use **events** instead of **binary sensors**.

In contrast, **events** are designed to capture **discrete actions**, not states. They're sent immediately and reliably, regardless of batching settings.

Using events can save you from subtle bugs caused by timing issues, especially as ESPHome evolves and changes its internals.

madbrain76, 2025-06-28:
> While events are the right choice for something like this as
> 
> > For me, it is a non-trivial amount of work to change from binary sensors to events.
> 
> Totally understandable — that’s a non-trivial amount of work, and I can see why the 100ms delay workaround makes sense for now.
> 
> If you haven't already, you could try lowering the [`batch_delay`](https://esphome.io/components/api.html#api-batch_delay) in the API component to something like `10ms` or `20ms`. That way, you still get the benefits of batching without adding too much latency.
> 
> That said, using binary sensors to represent events does carry some risk. It depends on specific timing, which can _easily shift_ due to batching, system load, the state of the transmit buffer, or changes in ESPHome’s behavior down the road. Events are more robust for capturing discrete actions, since they aren't affected by how and when state updates get sent.
> 
> So your current workaround is totally fair, but if you ever run into flakiness again, switching to events could help avoid some of those edge cases.
> 
> When working with rapid or one-shot triggers, such as button presses, motion activations, or other momentary inputs, more reliable to use **events** instead of **binary sensors**.
> 
> **events** are designed to capture **discrete actions**, not states. They're sent immediately and reliably, regardless of batching settings.

Thanks. I'll definitely consider changing to events.
I have not seen any flakiness since adding the 100ms delay. It is not noticeable. There is plenty of other latency in the system due to Wifi, especially my 222 Wifi smart bulbs, that is sometimes measured in seconds rather than milliseconds, if the bulbs roam to distant APs. I'm pushing the envelope of what's possible with Wifi in a single home, even a very large one with 9 APs - currently with a total of 304 Wifi clients, 297 of which are on 2.4 GHz. I'm not really sure how to solve this until some IOT devices start moving to 5 or 6 GHz, which have far more channels available than 2.4. Essentially no manufacturer is doing this yet. Of all my IOT devices, only my 3 Enphase solar gateways support both 2.4 and 5 GHz, and I set them up on 5 GHz already. The only other possibility I see is going back to dumb bulbs and using smart dimmer switches instead. It's a lot more expensive and labor intensive to replace 97 switches than 222 bulbs due to labor involved. Z-wave also has RF scalability problems with my current 75 nodes, sigh. Alternately, non-smart dimmer switches can be used, but that's not a great option either. I do need the dimming in order to have enough light with the macular degeneration when needed. I can't have full brightness everywhere at all times as it causes eyestrain, especially for others.

bdraco, 2025-06-28:
If you're sticking with binary sensors instead of switching to events, I'd recommend increasing the delay to at least 200ms. The 100ms delay doesn't guarantee the transmit buffer will have space available, which could still result in dropped messages. A 200ms delay should provide reliable operation in virtually all scenarios unless you have severe network issues.

The fundamental problem is that this component models discrete IR events as binary sensors, which is architecturally flawed. Binary sensors are designed for state changes, not instantaneous events. This design mismatch inevitably leads users to create configurations that appear to work initially but prove unreliable under real-world conditions.

madbrain76, 2025-07-02:
> If you're sticking with binary sensors instead of switching to events, I'd recommend increasing the delay to at least 200ms. The 100ms delay doesn't guarantee the transmit buffer will have space [...]

Thanks. I just changed it to 200ms

If the binary sensors are not suitable for IR events, why were these events made available that way ?

I may switch to events eventually when I have the time.

bdraco, 2025-07-02:
> If the binary sensors are not suitable for IR events, why were these events made available that way ?

I can’t say for certain why it was designed that way, but it was likely a case of misunderstanding or oversimplifying the use case. Modeling IR events as binary sensors might have seemed like a quick fit at the time, even though it doesn’t align well with how binary sensors are meant to be used.

ssieb, 2025-07-02:
They were added long before there were other options.

jkosharek, 2025-07-02:
For me it worked just fine before 2025.6.0 and binary_sensor is still in the IR Remote Receiver docs https://esphome.io/components/remote_receiver:
[image]

Not a single mention of events anywhere on the documents page. Seems this was pushed in 2025.6.0 without consideration of what it breaks or providing instructions on the best practice as it relates to IR Remote Receiver in ESPHome and HA.

jkosharek, 2025-07-02:
> > For me it worked just fine before 2025.6.0 and binary_sensor is still in the IR Remote Receiver docs [esphome.io/components/remote_receiver](https://esphome.io/components/remote_receiver): [image]
> > Not a single mention of events anywhere on the documents page. Seems this was pushed in 2025.6.0 without consideration of what it breaks or providing instructions on the best practice as it relates to IR Remote Receiver in ESPHome and HA.
> 
> Using `binary_sensor` with `remote_receiver` is still a valid use case for local-only setups. Unfortunately, we didn't realize this would become a problem because the underlying design limitations of `remote_receiver` weren’t fully on our radar when the `api` changes were made to fix other issues. We definitely should have caught this earlier and communicated the change better. We'll work on improving visibility into these kinds of interactions and make sure users aren't left without guidance in future updates.

I think some notes on the remote_receiver page would at least be helpful and a warning about binary_sensor are  essentially non-functional in HA, or at least that is how I understand it. Here is the abbreviated config I moved to, is this the best practice?
**### ESPHome:**
```
event:
  - platform: template
    name: Events
    id: events
    event_types:
      - Wash Back Normal

binary_sensor:
  - platform: remote_receiver
    id: back_normal
    receiver_id: remote_receiver_input
    toto:
      command: 0x80
      rc_code_1: 0x2
      rc_code_2: 0xC
    on_press:
      - event.trigger:
          id: events
          event_type: Wash Back Normal
```

**### HA Automation:**
```
triggers:
  - trigger: state
    entity_id:
      - event.toilet_ir_master_events
    attribute: event_type
    to: Wash Back Normal
    id: Master-Back-Normal
```
```
actions:
  - choose:
      - conditions:
          - condition: trigger
            id:
              - Master-Back-Normal
        sequence:
          - action: light.turn_on
            metadata: {}
            data: {}
            target:
              entity_id: light.master_toilet_fan_s1
```

glmnet, 2025-07-02:
I feel this is really creating an issue instead of fixing something else. Batch updates are great, e.g. if 10 binary sensors changes at the same time, send all the changes at the same time in 1 packet, but that does not mean to filter any data.
If an event is logged in esphome as a button (a binary sensor) being pressed, then there must consistently be an event in HA.
For binary sensors to not flood the network one should define properly debounce mechanisms e.g. using delayed_on_off, this will make the effect on esphome side match the effect on ha side.

@bdraco would it be too hard to actually have two features here? one that will batch updates and the other that will filter updates if the state change returned back to its last sent value?

bdraco, 2025-07-02:
> I feel this is really creating an issue instead of fixing something else. Batch updates are great, e.g. if 10 binary sensors changes at the same time, send all the changes at the same time in 1 [...]

We would need to track all of that in memory and with the limitations of esp8266 that would likely not be a good tradeoff.

bdraco, 2025-07-02:
https://github.com/esphome/esphome/pull/9298 is a possible solution, but as the PR states, its also a breaking change, and the tradeoffs may be unacceptable.

bdraco, 2025-07-03:
https://github.com/esphome/esphome/pull/9298 proposes to send states immediately when there is space available in the tx buffer by setting `batch_delay: 0`. This effectively restores the behavior pre 2025.6.0 on an opt-in basis for anyone who wants it

closed 2025-07-03 by bdraco