esphome/issues #7157: lvgl.qrcode.update does not work with WiFi strrings

agillis, 2025-06-23:
### The problem

This does does not work a creates a QRcode that returns gibberish. I can use lvgl.qrcode.update to generate a QRCode of anything else and it works fine. Only WiFi text strings seems to be broken.

```
  - platform: template
    name: "Device Name"
    id: device_name
    internal: true
    lambda: |-
      return {App.get_name().c_str()};
    on_value:
      then:
        - lvgl.label.update:
            id: device_name_label
            text:
              format: "%s"
              args: [ 'id(device_name).state.c_str()' ]
        - lvgl.qrcode.update:
            id: cp_login
            text: 
              format: "WIFI:S:%s;T:WPA;P:password;I:192.168.4.1;;"
              args: [ 'id(device_name).state.c_str()' ]
```

This works fine too.

```
              - qrcode:
                  id: cp_login
                  size: 100
                  text: "WIFI:S:nonet;T:WPA;P:password;I:192.168.4.1;;"  
```

### Which version of ESPHome has the issue?

2025.6.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

Latest

### What platform are you using?

ESP32-IDF

### Board

SDL

### Component causing the issue

lvgl.qrcode.update

### YAML Config

```yaml
- platform: template
    name: "Device Name"
    id: device_name
    internal: true
    lambda: |-
      return {App.get_name().c_str()};
    on_value:
      then:
        - lvgl.label.update:
            id: device_name_label
            text:
              format: "%s"
              args: [ 'id(device_name).state.c_str()' ]
        - lvgl.qrcode.update:
            id: cp_login
            text: 
              format: "WIFI:S:%s;T:WPA;P:password;I:192.168.4.1;;"
              args: [ 'id(device_name).state.c_str()' ]
```

### Anything in the logs that might be useful for us?

```txt

```

### Additional information

[image]
This is the generated Qcode

ssieb, 2025-06-23:
What if you provide a static string as an argument or use a format with no arguments?

agillis, 2025-06-23:
This does product the same (bad) result.

```
        - lvgl.qrcode.update:
            id: cp_login
            text:  "WIFI:S:mynetwork;T:WPA;P:password;I:192.168.4.1;;"
```

agillis, 2025-06-23:
I have tried a lot of different iterations anything WIFI:S:xxxxxxxx;T:WPA;P:xxxxxxxx;; seems to kill lvgl.qrcode.update. Again this is only for **lvgl.qrcode.update**

```              - qrcode:
                  id: cp_login
                  size: 100
                  text: "WIFI:S:nonet;T:WPA;P:password;;" 
```

Has no issues at all.

ssieb, 2025-06-23:
Does anything work for update?  I was assuming it is just that update is broken somehow, not specific strings.

agillis, 2025-06-23:
Yes This works great

```
        - lvgl.qrcode.update:
            id: cp_login
            text:  "http://google.com"
```

As does any other string I can think of.

agillis, 2025-06-23:
only WIFI:S:xxxxxxxx;T:WPA;P:xxxxxxxx;;  seems to stop it from working.

ssieb, 2025-06-23:
Could it be length?

ssieb, 2025-06-23:
Ok, that's weird.  I've let Clyde know.

closed 2025-06-23 by clydebarrow

agillis, 2025-06-24:
That works much better! Thanks Clyde