esphome/issues #7178: WiFi.h not found - Adafruit Feather Huzzah

brauhausdc, 2025-06-30:
### The problem

Getting the following error when trying to build/install for a couple Adafruit Feather Huzzah boards: 

INFO ESPHome 2025.7.0-dev
INFO Reading configuration /config/huzzah.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing huzzah (board: huzzah; framework: arduino; platform: platformio/espressif8266@4.2.1)
--------------------------------------------------------------------------------
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
Dependency Graph
|-- ESPAsyncTCP @ 2.0.0
|-- ESP8266WiFi @ 1.0
|-- ESPAsyncWebServer @ 3.7.8
|-- DNSServer @ 1.1.1
|-- ESP8266mDNS @ 1.2
|-- noise-c @ 0.1.6
|-- Wire @ 1.0
|-- ArduinoJson @ 6.18.5
Compiling .pioenvs/huzzah/src/esphome/components/adc/adc_sensor_common.cpp.o
Compiling .pioenvs/huzzah/src/esphome/components/adc/adc_sensor_esp32.cpp.o
Compiling .pioenvs/huzzah/src/esphome/components/adc/adc_sensor_esp8266.cpp.o
Compiling .pioenvs/huzzah/src/esphome/components/adc/adc_sensor_libretiny.cpp.o
Compiling .pioenvs/huzzah/src/esphome/components/adc/adc_sensor_rp2040.cpp.o
Compiling .pioenvs/huzzah/src/esphome/components/api/api_connection.cpp.o
Compiling .pioenvs/huzzah/src/esphome/components/api/api_frame_helper.cpp.o
Compiling .pioenvs/huzzah/src/esphome/components/api/api_pb2.cpp.o
Compiling .pioenvs/huzzah/src/esphome/components/api/api_pb2_service.cpp.o
Compiling .pioenvs/huzzah/src/esphome/components/api/api_server.cpp.o
Compiling .pioenvs/huzzah/src/esphome/components/api/list_entities.cpp.o
Compiling .pioenvs/huzzah/src/esphome/components/api/proto.cpp.o
Compiling .pioenvs/huzzah/src/esphome/components/api/subscribe_state.cpp.o
Compiling .pioenvs/huzzah/src/esphome/components/api/user_services.cpp.o
Compiling .pioenvs/huzzah/src/esphome/components/mdns/mdns_component.cpp.o
Compiling .pioenvs/huzzah/src/esphome/components/pmsa003i/pmsa003i.cpp.o
Compiling .pioenvs/huzzah/src/esphome/components/status/status_binary_sensor.cpp.o
Compiling .pioenvs/huzzah/src/esphome/components/web_server/list_entities.cpp.o
Compiling .pioenvs/huzzah/src/esphome/components/web_server/web_server.cpp.o
Compiling .pioenvs/huzzah/src/esphome/components/web_server/web_server_v1.cpp.o
Compiling .pioenvs/huzzah/src/esphome/core/application.cpp.o
Compiling .pioenvs/huzzah/src/esphome/core/component.cpp.o
Compiling .pioenvs/huzzah/src/esphome/core/component_iterator.cpp.o
Compiling .pioenvs/huzzah/src/esphome/core/util.cpp.o
Compiling .pioenvs/huzzah/src/main.cpp.o
Archiving .pioenvs/huzzah/libfb1/libESP8266WiFi.a
Archiving .pioenvs/huzzah/lib181/libHash.a
Compiling .pioenvs/huzzah/lib46e/RPAsyncTCP/AsyncPrinter.cpp.o
In file included from /root/.platformio/lib/RPAsyncTCP/src/AsyncPrinter.h:26,
                 from /root/.platformio/lib/RPAsyncTCP/src/AsyncPrinter.cpp:26:
/root/.platformio/lib/RPAsyncTCP/src/RPAsyncTCP.h:5:10: fatal error: WiFi.h: No such file or directory

**************************************************************
* Looking for WiFi.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:WiFi.h"
* Web  > https://registry.platformio.org/search?q=header:WiFi.h
*
**************************************************************

    5 | #include <WiFi.h>
      |          ^~~~~~~~
compilation terminated.
*** [.pioenvs/huzzah/lib46e/RPAsyncTCP/AsyncPrinter.cpp.o] Error 1
========================= [FAILED] Took 25.68 seconds =========================

### Which version of ESPHome has the issue?

2025.7.0-dev

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

2025.6.3 - but this isn't HA specific. 

### What platform are you using?

ESP8266

### Board

Adafruit Feather Huzzah

### Component causing the issue

AsyncPrinter

### YAML Config

```yaml
esphome:
  name: huzzah
  comment: Front Porch
  area: Front Porch

esp8266:
  board: huzzah

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: !secret encryption_key 
  # password: !secret api_password

ota:
  platform: esphome
  password: !secret ota_password
  port: 8266

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Huzzah Fallback Hotspot"
    password: "XXXXXXXXXXXXXXXX"
  
  manual_ip:
    # Set this to the IP of the ESP
    static_ip: 192.168.AA.BB
    # Set this to the IP address of the router. Often ends with .1
    gateway: 192.168.AA.CC
    # The subnet of the network. 255.255.255.0 works for most home networks.
    subnet: 255.255.255.0
    

captive_portal:

switch:
  - platform: gpio
    name: "Huzzah Red LED"
    pin: 0
    inverted: True
  - platform: gpio
    name: "Huzzah Blue LED"
    pin: 2
    inverted: True
    

# Create BME Sensor on I2C
i2c:
  sda: SDA
  scl: SCL
  scan: True

sensor:

  - platform: uptime
    name: "H1 Uptime Sensor"
    
  - platform: adc
    pin: VCC
    name: "H1 VCC Voltage"

  - platform: wifi_signal
    name: "H1 WiFi Signal Sensor"
    update_interval: 60s
  

  - platform: bme280_i2c
    update_interval: 60s
    address: 0x77
    temperature:
      name: "BME280 Temperature"
      id: bme280_temperature
      filters:
        - offset: 0.00
      force_update: true
      accuracy_decimals: 2
    pressure:
      name: "BME280 Pressure"
      id: bme280_pressure
    humidity:
      name: "BME280 Relative Humidity"
      id: bme280_humidity
      force_update: true
      accuracy_decimals: 2

  - platform: pmsa003i
    address: 0x12
    pm_1_0:
      name: "PM1.0"
    pm_2_5:
      name: "PM2.5"
    pm_10_0:
      name: "PM10.0"
    pmc_0_3:
      name: "PMC <0.3µm"
    pmc_0_5:
      name: "PMC <0.5µm"
    pmc_1_0:
      name: "PMC <1µm"
    pmc_2_5:
      name: "PMC <2.5µm"
    pmc_5_0:
      name: "PMC <5µm"
    pmc_10_0:
      name: "PMC <10µm"
      

binary_sensor:
    - platform: status
      name: "BME280 Status"
      id: bme280_status
      
    
# Example configuration entry
web_server:

# Activates prometheus /metrics endpoint
prometheus:
```

### Anything in the logs that might be useful for us?

```txt
Compile log posted above.
```

randybb, 2025-06-30:
Clear build files as it is probably using something that is not valid anymore

ssieb, 2025-07-05:
This is not a general issue.  It's something specific to your environment.  Try re-installing the addon or docker container or whatever you're using.

rotary65, 2025-07-05:
This was verified on on two separate platforms: ESPHome add on on HAOS on  mini PC, the other on a Pi5 running ESPHome on command line. Both cases with a full config and basic configuration with  identical results. Please explain how that can be environmental?

ssieb, 2025-07-05:
Primarily because no one else is seeing this.  It's basic functionality that would be very obvious if it was actually broken.  You also mention having download issues.

rotary65, 2025-07-05:
> Primarily because no one else is seeing this. It's basic functionality that would be very obvious if it was actually broken. You also mention having download issues.

A more standard YAML for another device on the same ESPHome add-on (like a basic sensor or power meter) compiles fine because it only uses the built-in ESPHome components that come pre-installed and are already set up in the add-on environment.

The Ratgdo YAML, on the other hand, uses an external component (the ratgdo library from GitHub). This requires ESPHome and PlatformIO to download and install extra code and libraries during the build process. If there’s any issue with downloading, installing, or linking these extra files—such as a network, SSL, or packaging problem—the build can fail with errors like missing WiFi.h.

In short:

- Standard YAMLs only use what’s already there, so they’re less likely to run into problems.
- YAMLs with external components depend on extra downloads and setup, which can fail if there’s any issue with the toolchain or network.

That’s why a basic device YAML compiles, but the Ratgdo one does not, even though both are on the same ESPHome add-on.

brauhausdc, 2025-07-05:
You have at least two different users seeing this issue. I’m seeing it with Feather Huzzah.Platform here is esphome running in docker on X86-64. Have cleaned build files. Still broken. Sent from my

ssieb, 2025-07-05:
@rotary65 if you're having a network problem, how is that the responsibility of esphome?

@brauhausdc did you try rebuilding the docker container?  Particularly, you need to clear the platformio cache.

rotary65, 2025-07-05:
> [@rotary65](https://github.com/rotary65) if you're having a network problem, how is that the responsibility of esphome?
> 
> [@brauhausdc](https://github.com/brauhausdc) did you try rebuilding the docker container? Particularly, you need to clear the platformio cache.

Thanks for your help—you're right, this isn't an ESPHome issue. After looking into it further, it turns out the problem is with PlatformIO itself, which isn't downloading or setting up the ESP32 Arduino framework correctly. That's why files like `WiFi.h` are missing during the build. I'll report this to the PlatformIO team instead, since it's a technical issue on their end. Appreciate the guidance!

ssieb, 2025-07-05:
@rotary65 but why is it only happening on your systems?

rotary65, 2025-07-05:
> [@rotary65](https://github.com/rotary65) but why is it only happening on your systems?

The reason this issue is probably showing up for me (and not everyone) is because it depends on a few factors outside of ESPHome itself:

PlatformIO uses different download servers (mirrors) around the world. If a particular server has a problem (like a misconfigured security setting), only people routed to that server will see errors like missing WiFi.h or failed downloads.

If someone already has the needed files cached from before, they won’t see the problem. But if you’re doing a fresh install or need to download the framework again, you’re more likely to hit the error.

It’s more likely to happen with projects that use external components (like ratgdo), because they often trigger a fresh download of the ESP32 Arduino framework.

So, it’s not just my system—it’s likely a PlatformIO technical issue that only affects certain users, depending on which server they’re connected to and whether they need to download new files. That’s why it looks random, but it’s probably actually a broader problem that just doesn’t hit everyone at the same time.

ssieb, 2025-07-05:
esphome doesn't have any way to verify that.

rotary65, 2025-07-05:
> esphome doesn't have any way to verify that.

Yes, so in my case, my next step would probably be to follow up with the ratdgo project on github

brauhausdc, 2025-07-09:
One question, and I recognize this may still be an upstream problem — I note that this error happens when it’s trying to compile “RPAsyncTCP” - which happens a few lines after it successfully compiles ESPAsyncTCP. Why is it bothering with compiling the “RP” version, which I assume is intended for the RP2040 boards, when this is an ESP8266 based board?

ssieb, 2025-07-09:
There's some sort of dependency issue happening.  It's being investigated.

rotary65, 2025-07-15:
In my case, the issue turned out to be a missing wifi section in the device YAML configuration file. Once that was added, it compiled. So, it was my own mistake.

closed 2025-11-30 by github-actions[bot]