esphome/issues #7179: Fatal compile error: binary_sensor/binary_sensor.h missing when compiling as3935_spi component on ESP32

21BakerAlfa, 2025-06-30:
### The problem

Using versions of Home Assistant since 2018
CURRENT Setup:
 Home Assistant OS
Core 2025.6.3
Supervisor 2025.06.2
Operating System 15.2
Frontend 20250531.4
ESPHome Device Builder
Current version: 2025.6.2
Raspberry pi 4
New, fresh install on new SD card June 27, 2025
------I’m experiencing a compile failure when building an ESPHome project using the official as3935_spi component on an ESP32.  AS3935 module with ESP32 dev board
This same device and YAML compiled and worked perfectly in 2022. After a fresh Home Assistant install in June 2025, I now get:   "In file included from src/esphome/components/as3935_spi/as3935_spi.cpp:1:
src/esphome/components/as3935_spi/as3935_spi.h:7:10: fatal error: esphome/components/binary_sensor/binary_sensor.h: No such file or directory
 #include "esphome/components/binary_sensor/binary_sensor.h"

Steps tried:

    1-Delete ALL Lightning Detector YAML files
    2-Re-Wrote New YAML file
    3-Clean build in ESPHome
    4-Stripped YAML to Minimal YAML with only basic components
    5-No custom components present
    6-Completely RE-Install Home Assistant AGAIN on ANOTHER new SD card
    7-Repeated steps 1 through 5
    8-Double checked to make sure the AS3935 is still a valid component for ESPHome
    9-NOTE: Compiled several other revised YAML files for other devices and reloaded onto their respective boards [ESP32 and 8266] ALL OK, no errors/failures.

    Issue persists even with new, stripped down YAML

I checked the config/esphome/ folder and there’s no leftover custom code. It seems the binary_sensor.h header is missing from the ESPHome installation. I suspect a missing file in the Docker image or PlatformIO environment.

### Which version of ESPHome has the issue?

ESPHome Device Builder Current version: 2025.6.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

 Home Assistant OS Core 2025.6.3 Supervisor 2025.06.2 Operating System 15.2 Frontend 20250531.4

### What platform are you using?

ESP32

### Board

ESP32 Wroom Dev Board with AS3935 nodule

### Component causing the issue

AS3935 nodule

### YAML Config

```yaml
[Stripped down YAML]
esphome:
  name: lightning-detector
  platform: ESP32
  board: esp32dev

wifi:
  ssid: "YourSSID"
  password: "YourPassword"

logger:
api:
ota:

spi:
  clk_pin: GPIO18
  miso_pin: GPIO19
  mosi_pin: GPIO23

as3935_spi:
  irq_pin: GPIO25
  cs_pin: GPIO5
```

### Anything in the logs that might be useful for us?

```txt
INSTALL ATTEMPT LOGS:
INFO ESPHome 2025.6.2
INFO Reading configuration /config/esphome/lightning-detector.yaml...
WARNING GPIO5 is a strapping PIN and should only be used for I/O with care.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
INFO Generating C++ source...
INFO Compiling app...
Processing lightning-detector (board: esp32dev; framework: arduino; platform: platformio/espressif32@5.4.0)
--------------------------------------------------------------------------------
Library Manager: Installing esphome/AsyncTCP-esphome @ 2.1.4
INFO Installing esphome/AsyncTCP-esphome @ 2.1.4
Unpacking  [####################################]  100%
Library Manager: AsyncTCP-esphome@2.1.4 has been installed!
INFO AsyncTCP-esphome@2.1.4 has been installed!
Library Manager: Installing esphome/ESPAsyncWebServer-esphome @ 3.3.0
INFO Installing esphome/ESPAsyncWebServer-esphome @ 3.3.0
Unpacking  [####################################]  100%
Library Manager: ESPAsyncWebServer-esphome@3.3.0 has been installed!
INFO ESPAsyncWebServer-esphome@3.3.0 has been installed!
Library Manager: Resolving dependencies...
INFO Resolving dependencies...
Library Manager: Installing bblanchon/ArduinoJson @ 6.18.5
INFO Installing bblanchon/ArduinoJson @ 6.18.5
Unpacking  [####################################]  100%
Library Manager: ArduinoJson@6.18.5 has been installed!
INFO ArduinoJson@6.18.5 has been installed!
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
Dependency Graph
|-- AsyncTCP-esphome @ 2.1.4
|-- WiFi @ 2.0.0
|-- FS @ 2.0.0
|-- Update @ 2.0.0
|-- ESPAsyncWebServer-esphome @ 3.3.0
|-- DNSServer @ 2.0.0
|-- ESPmDNS @ 2.0.0
|-- SPI @ 2.0.0
|-- ArduinoJson @ 6.18.5
Compiling .pioenvs/lightning-detector/src/esphome/components/api/api_connection.cpp.o
Compiling .pioenvs/lightning-detector/src/esphome/components/api/api_frame_helper.cpp.o
Compiling .pioenvs/lightning-detector/src/esphome/components/api/api_pb2.cpp.o
Compiling .pioenvs/lightning-detector/src/esphome/components/api/api_pb2_service.cpp.o
Compiling .pioenvs/lightning-detector/src/esphome/components/api/api_server.cpp.o
Compiling .pioenvs/lightning-detector/src/esphome/components/api/list_entities.cpp.o
Compiling .pioenvs/lightning-detector/src/esphome/components/api/proto.cpp.o
Compiling .pioenvs/lightning-detector/src/esphome/components/api/subscribe_state.cpp.o
Compiling .pioenvs/lightning-detector/src/esphome/components/api/user_services.cpp.o
Compiling .pioenvs/lightning-detector/src/esphome/components/as3935/as3935.cpp.o
Compiling .pioenvs/lightning-detector/src/esphome/components/as3935_spi/as3935_spi.cpp.o
Compiling .pioenvs/lightning-detector/src/esphome/components/captive_portal/captive_portal.cpp.o
In file included from src/esphome/components/as3935_spi/as3935_spi.cpp:1:
src/esphome/components/as3935_spi/as3935_spi.h:6:10: fatal error: esphome/components/sensor/sensor.h: No such file or directory
 #include "esphome/components/sensor/sensor.h"
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
*** [.pioenvs/lightning-detector/src/esphome/components/as3935_spi/as3935_spi.cpp.o] Error 1
========================= [FAILED] Took 31.30 seconds =========================
```

### Additional information

I am not an experienced 'coder', I plod along as best I can but normally can [eventually] work out the problem.  This one is simply beyond my limited capabilities.  Any suggestions would be appreciated.
I sincerely appreciate all the hard work you all have done to produce a consistently superior system that everyone can use.
Thank you

ssieb, 2025-07-15:
That appears to be a bug in the component.  It assumes that you will configure the binary sensor in the config.  Right now, the component is doing nothing, because you haven't configured anything to get data from it.

Pierre-Rabu, 2025-07-15:
Thanks ^^

i add :
binary_sensor:
  - platform: as3935
    name: "SensGarden Détection Foudre"
( thanks Gemini to ) 

and he work 
[image]