esphome/issues #7182: Data corruption in BluetoothLERawAdvertisementsResponse/BluetoothLERawAdvertisement/Data

Ga-Ol-St, 2025-07-01:
### The problem

Did try to setup BLE Battery management system via Bluetooth Proxy, hit issue that component in HomeAssistant don't recognize Service UUID which was delivered from EspHome.

In particular HomeAssistant Advertising Monitor reports that BLE device have Service UUID "0000fee7ffe0-0000-1000-8000-00805f9b34fb" - which is corrupted 128Bit UUID (first portion should have 4 byte but have 6) 

Enabled VERY_VERBOSE logging in ESPHome and analyzed log, spot issue in sending of "bluetooth_le_raw_advertisements_response" BluetoothLERawAdvertisement.data doesn`t match with data from "esp32_ble_tracker"

Screenshot (to visualize unprintable characters and highlight corruption)

[image]

P.S. also i see corruption on other devices types, for example in the logs i see different corruption behavior in device with MAC: A4:C1:38:B6:63:C9, Advertisement data have 30 bytes but corresponding BluetoothLERawAdvertisement.data have only approx 7 bytes, corresponding BluetoothLERawAdvertisement.address: 181149788160386

### Which version of ESPHome has the issue?

2025.6.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2025.6.3

### What platform are you using?

ESP32-IDF

### Board

ESP32-WROOM

### Component causing the issue

Bluetooth

### YAML Config

```yaml
substitutions:
  name: esphome-web-3a9a0c
  friendly_name: BTProxy 3a9a0c Majsternja

esphome:
  name: ${name}
  friendly_name: ${friendly_name}
  min_version: 2024.6.0
  name_add_mac_suffix: false
  project:
    name: esphome.web
    version: dev

esp32:
  board: esp32dev
  framework:
    type: esp-idf

# Enable logging
logger:
   level: VERY_VERBOSE

# Enable Home Assistant API
api:

# Allow Over-The-Air updates
ota:
- platform: esphome

[... 122 lines cut ...]
[13:19:01][VV][api.service:013]:   address_type: 0
[13:19:01][VV][api.service:013]:   data: '\xe0\xff\xe7\xfe\xffe\x88\xa0\xc8G\x8c\xea\xd1\xc1	M2_B1A8S10P'
[13:19:01]}
[13:19:01][VV][api.service:013]:   advertisements: BluetoothLERawAdvertisement {
[13:19:01][VV][api.service:013]:   address: 181149788160386
[13:19:01][VV][api.service:013]:   rssi: -85
[13:19:01][VV][api.service:013]:   address_type: 0
[13:19:01][VV][api.service:013]:   data: '\xd2\xfc@
[13:19:01][VV][esp32_ble_tracker:536]: Parse Result:
[13:19:01][VV][esp32_ble_tracker:555]:   Address: C8:47:8C:EA:D1:C1 (PUBLIC)
[13:19:01][VV][esp32_ble_tracker:558]:   RSSI: -71
[13:19:01][VV][esp32_ble_tracker:559]:   Name: 'M2_B1A8S10P'
[13:19:01][VV][esp32_ble_tracker:567]:   Ad Flag: 6
[13:19:01][VV][esp32_ble_tracker:570]:   Service UUID: 0xFFE0
[13:19:01][VV][esp32_ble_tracker:570]:   Service UUID: 0xFEE7
[13:19:01][VV][esp32_ble_tracker:581]:   Manufacturer ID: 0x0B65, data: 88.A0.C8.47.8C.EA.D1.C1 (8)
[13:19:01][VV][esp32_ble_tracker:591]:   Adv data: 02.01.06.05.02.E0.FF.E7.FE.0B.FF.65.0B.88.A0.C8.47.8C.EA.D1.C1.0C.09.4D.32.5F.42.31.41.38.53.31.30.50 (34)
```

### Additional information

full log file (including build and deploy process)
P.S. i have many Bluetooth devices so it's full of traces, but it's in original state with all data.

[logs_esphome-web-3a9a0c_run (4).txt](https://github.com/user-attachments/files/20997852/logs_esphome-web-3a9a0c_run.4.txt)

jesserockz, 2025-07-01:
Hi @Ga-Ol-St,

Are you able to test using https://github.com/esphome/esphome/pull/9288 ?

You would have to install it manually as it cannot be used as an external_component.

That will make it easier to read the raw advertisement data logs to compare

Jesse

bdraco, 2025-07-02:
I wonder if we are encoding as string when it should be bytes

bdraco, 2025-07-02:
So it probably drops everything after the null

bdraco, 2025-07-02:
can you try https://github.com/esphome/esphome/pull/9289 ?

Ga-Ol-St, 2025-07-02:
@bdraco 
Sure, here are results from "fix_bytes_encoding" branch:

now ESPHome logs looks like this:

[image]

Corresponding log from HomeAssistant look like this:
```txt
2025-07-02 04:40:58.060 DEBUG (MainThread) [aioesphomeapi.connection] esphome-web-3a9a0c @ 10.168.192.233: Got message of type BluetoothLERawAdvertisementsResponse: advertisements {
  address: 220209632432577
  rssi: -94
  data: "\002\001\006\005\002\340\377\347\376\013\377e\013\210\240\310G\214\352\321\301\014\tM2_B1A8S10P"
}```

Ga-Ol-St, 2025-07-02:
Same "e" and "G" characters are still there :-)

Ga-Ol-St, 2025-07-02:
Just a hint, it's possible that those extra characters are present in esp32_ble_tracker - but we don't see them as logs display data in HEX ?

Ga-Ol-St, 2025-07-02:
Sorry wrong device in previous nessage.
{"name":"M2_B1A8S10P","address":"C8:47:8C:EA:D1:C1","rssi":-98,"manufacturer_data":{"2917":"88a0c8478cead1c1"},"service_data":{},"service_uuids":["0000fee7ffe0-0000-1000-8000-00805f9b34fb"],"source":"F8:B3:B7:3A:9A:0E","connectable":true,"time":1751423297.2373955,"tx_power":null}

bdraco, 2025-07-02:
if you turn on bluetooth debug logging in HA can you see the raw data coming across correctly ?

bdraco, 2025-07-02:
I'm starting to think the problem isn't with ESPHome at all

Ga-Ol-St, 2025-07-02:
> if you turn on bluetooth debug logging in HA can you see the raw data coming across correctly ?

No, that was a weird in HA, when debug logging enabled for Bluetooth in HA those devices never appeared in logs, i did see some other devices but not those two. It was weird how Adv.Monitor see them but debug never show them.
I was able to detect them only whe start debug in Esphome component thru UI and found just that advertising messages.

bdraco, 2025-07-02:
It looks like a bug in the parser and not ESPHome https://github.com/Bluetooth-Devices/bluetooth-data-tools/pull/155

bdraco, 2025-07-02:
### Structure 2: Incomplete List of 16-bit Service UUIDs
- **Length**: 0x05 (5 bytes follow)
- **AD Type**: 0x02 (Incomplete List of 16-bit Service Class UUIDs)
- **Data**: 0xE0 0xFF 0xE7 0xFE
  - UUID 1: 0xFFE0 (reversed byte order in BLE)
  - UUID 2: 0xFEE7 (reversed byte order in BLE)
  - **Meaning**: Device advertises two 16-bit service UUIDs

The problem is the parser only supports reading one value an in this case two values are encoded in the structure

Ga-Ol-St, 2025-07-02:
There was some reports that when somebody did use bluetooth directly in HA everything was fine, but when did use Bluetooth proxy then issue appeared

bdraco, 2025-07-02:
The matches what I'm seeing in the parser.

We need to add support for multiple 16-bit/32-bit uuids in the same record to the parser

bdraco, 2025-07-02:
I'm pretty sure there is no data corruption in ESPHome. The problem is the lack of support in the parser

bdraco, 2025-07-02:
Actually the parser doesn't support 32-bit uuids at all so we need to add support for them as well

Ga-Ol-St, 2025-07-02:
Well, i did see some discrepancy in logs an some extra characters in HA and in EspHome so opened a bug, if you say that it's a missing support of some cases - then i'm fine with that.

Just hope this issue will not stuck in this state forever.

Thanks for your time and help.

jesserockz, 2025-07-02:
> Same "e" and "G" characters are still there :-)

For future reference, 'e' is `65` in hex and 'g' is `47`. It just so happens that those hex numbers are in the range of the alphabet and can be logged as such by the python logger on the aioesphomeapi/HA side

bdraco, 2025-07-02:
> [@bdraco](https://github.com/bdraco) won't the change [home-assistant/core#147920](https://github.com/home-assistant/core/pull/147920) be a bit limited in the issue? e.g. [home-assistant/core@`afb247c`/homeassistant/loader.py#L179](https://github.com/home-assistant/core/blob/afb247c90723b59b3d8f52e499fd8db986817b44/homeassistant/loader.py#L179) the matcher data structure only takes a string, i.e. one UUID in my opinion Thus, I'm not able to tell the Bluetooth integration to only match devices that announce two UUIDs, right? That limits my auto-detection feature a lot as most BMS manufacturers use 0xFFF0 or 0xFF00 and don't give a usable manufacturer ID. 😞

That PR was specifically focused on fixing the originally reported parser issue. Supporting matching on multiple UUIDs in a single advertisement is outside its scope and would be considered a feature request.

Matching on both UUIDs would mostly be an optimization to avoid invoking the callback unnecessarily. But in practice, that’s likely a rare use case. It’s generally more efficient to match on either UUID and handle the filtering yourself in the callback, since most use cases don’t require that level of specificity. Adding support for compound UUID matching would increase the complexity of the matcher and could introduce a performance hit overall.