esphome/issues #7200: esphome 2025.6.2 and openthread component

jeffbrown57, 2025-07-05:
### The problem

Can i assume this is really alpha level software and the 2 days i've spent trying to get it working with
either apple homepod as TBR or espressif thread border router kit as TBR. Should i just wait a few months
for this to mature ??? just adding openthread component with ( what i assume to be valid values for channel, panid, extpanid , networkkey) results in either panics , or never getting out of bootloader. Removing the openthread: component revives the esp32c6 working again ( i can see logs)

### Which version of ESPHome has the issue?

2025.6.2

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

2025.4.3

### What platform are you using?

ESP32-IDF

### Board

esp32-c6-devkitm-1

### Component causing the issue

openthread

### YAML Config

```yaml
esphome:
  name: c6thr-weather-station
  friendly_name: c6thr-weather-station
  platformio_options:
    build_flags:
      -DCONFIG_OPENTHREAD_MTD=1
      -DCONFIG_OPENTHREAD_FTD=0
      -DCONFIG_OPENTHREAD_JOINER=1
      -DCONFIG_NIMBLE_ENABLED=0
      -DCONFIG_LOG_DEFAULT_LEVEL=3  # 3 = INFO

esp32:
  board: esp32-c6-devkitm-1
  framework:
    type: esp-idf

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "Nnh6CqjYC42nHEODql5zgieNtiBg99+tPhyrjxq0KSs="

ota:
  - platform: esphome
    password: "6c11df78387331845fd6e30b7154c838"

network:
  enable_ipv6: True

openthread:
  network_name: "OpenThread-9873"
  channel: 17
  pan_id: 0x0298
  ext_pan_id: 0x330d857354330133
  network_key: 0x1035060004001fffe00208e227ac6a7
  pskc: 0x5e1fd7ae81a910c0402a0f7f8
  
#wifi:
# ssid: !secret wifi_ssid
 # password: !secret wifi_password

 # manual_ip: 
 #   gateway: 192.168.1.1
 #   static_ip: 192.168.1.119
 #   subnet: 255.255.255.0
 #   dns1: 192.168.1.1

i2c:
  - scan: False
    scl: GPIO7
    scl_pullup_enabled: False
    sda: GPIO6
    sda_pullup_enabled: False
    frequency: 100khz
    id: bus_a

sensor:
  - platform: bme280_i2c
    temperature:
      name: "BME280 Temperature"
    pressure:
      name: "BME280 Pressure"
    humidity:
      name: "BME280 Humidity"
    address: 0x76  
    i2c_id: bus_a
```

### Anything in the logs that might be useful for us?

```txt
mostly just bootloader messages as the openthread component kills the esp32c6
```

sadpixel, 2025-07-07:
Switch to the dev release of ESPHome, which has already merged some OpenThread-related fixes. I had the same issue as you with bootloader crashes, which were caused by the TLV parser tripping up. (The dev release uses Espressif’s own built-in TLV parser, whereas the current stable release uses a home-rolled solution that has issues.)

Drop all the platformio **build_flags** options – they’re not needed and may cause more harm than good. All you need to add for Thread is this:

```yaml
network:
  enable_ipv6: true

openthread:
  tlv: ABC123
  # device_type: MTD
```

Assuming you have a Thread section in your Home Assistant install, grab your TLV as spat out by Home Assistant under Settings > Devices/Services > Thread > Config Gear > Info button > "Active dataset TLVs", and you should be in business. If you can grab the TLV from somewhere else, great. If you're planning to provide each field individually as you are doing now, you'll have to use the **force_dataset: _true_** option as well.

Device type defaults to Full Thread Device (FTD), aka a router node in the mesh, but you can disable mesh routing by setting it to a Minimal Thread Device (MTD).

You can get debug info about the Thread stack running on your C6 like so:

```yaml
text_sensor:
  - platform: openthread_info
    ip_address:
      name: "Off-mesh routable IP Address"
    channel:
      name: "Channel"
    role:
      name: "Device Role"
    rloc16:
      name: "RLOC16"
    ext_addr:
      name: "Extended Address"
    eui64:
      name: "EUI64 Interface ID"
    network_name:
      name: "Network Name"
    network_key:
      name: "Network Key"
    pan_id:
      name: "PAN ID"
    ext_pan_id:
      name: "Extended PAN ID"
```
(Note to devs: This is **_NOT_** documented, but it should be!!)

The OpenThread stack is not "alpha", but it _is_ brand-new to ESPHome, so it'll take a moment to work out the kinks and optimize. For example, you can't yet stop and restart the Thread network at runtime as you can with WiFi, nor can you re-associate to a different mesh at runtime. That said, performance for me is instant and reliable.

Do note that I've found my C6s run several degrees hotter in Thread mode than WiFi 6 mode – not an _insane_ amount hotter, but hotter nonetheless. We'll see if this improves as the integration with ESPHome improves...

jeffbrown57, 2025-07-07:
> Switch to the dev release of ESPHome, which has already merged some OpenThread-related fixes. I had the same issue as you with bootloader crashes, which were caused by the TLV parser tripping up. (The dev release uses Espressif’s own built-in TLV parser, whereas the current stable release uses a home-rolled solution that has issues.)
> 
> Drop all the platformio **build_flags** options – they’re not needed and may cause more harm than good. All you need to add for Thread is this:
> 
> network:
>   enable_ipv6: true
> 
> openthread:
>   tlv: ABC123
>   # device_type: MTD
> Assuming you have a Thread section in your Home Assistant install, grab your TLV as spat out by Home Assistant under Settings > Devices/Services > Thread > Config Gear > Info button > "Active dataset TLVs", and you should be in business. If you can grab the TLV from somewhere else, great. If you're planning to provide each field individually as you are doing now, you'll have to use the **force_dataset: _true_** option as well.
> 
> Device type defaults to Full Thread Device (FTD), aka a router node in the mesh, but you can disable mesh routing by setting it to a Minimal Thread Device (MTD).
> 
> You can get debug info about the Thread stack running on your C6 like so:
> 
> text_sensor:
>   - platform: openthread_info
>     ip_address:
>       name: "Off-mesh routable IP Address"
>     channel:
>       name: "Channel"
>     role:
>       name: "Device Role"
>     rloc16:
>       name: "RLOC16"
>     ext_addr:
>       name: "Extended Address"
>     eui64:
>       name: "EUI64 Interface ID"
>     network_name:
>       name: "Network Name"
>     network_key:
>       name: "Network Key"
>     pan_id:
>       name: "PAN ID"
>     ext_pan_id:
>       name: "Extended PAN ID"
> (Note to devs: This is **_NOT_** documented, but it should be!!)
> 
> The OpenThread stack is not "alpha", but it _is_ brand-new to ESPHome, so it'll take a moment to work out the kinks and optimize. For example, you can't yet stop and restart the Thread network at runtime as you can with WiFi, nor can you re-associate to a different mesh at runtime. That said, performance for me is instant and reliable.
> 
> Do note that I've found my C6s run several degrees hotter in Thread mode than WiFi 6 mode – not an _insane_ amount hotter, but hotter nonetheless. We'll see if this improves as the integration with ESPHome improves...

Is the "dev" release just 2025.6.3.  ?

jeffbrown57, 2025-07-08:
> Nope. If you’re installing from Home Assistant, the Add-On Store has three separate releases listed: ESPHome, ESPHome Beta, and ESPHome Dev. You can install them side-by-side, and only compile using the dev branch specifically for your C6, while leaving your other nodes (if you have any) on the release version, which as of this writing is 2025.6.3.
> 
> The OpenThread fixes will eventually trickle from the dev release to the stable version — maybe as soon as 2025.7.0 is released, maybe a bit later.

OMG , thanks so much. I use Truenas Scale with docker so i pulled esphome/esphome:dev and updated the container rebuilt the firmware and it works !!!   it's 2025.7.0-dev  . Thanks for the addiional template sensor