esphome/issues #7201: Sonoff TH10/16 SI7021 dht [dht:051]: Invalid readings!

mxtreme311, 2025-07-05:
### The problem

Upgraded multiple devices from 2025.5.2 -> 2025.6.3 and all devices are reporting this error:

[dht:051]: Invalid readings! Check pin number and pull-up resistor and try manually specifying the model.

Nothing changed in the configs, these sensors are hardwired to GPIO14 (from my research) and the pull-up resistor is built in to the sensor from what I can see. I didn't see anything in the release notes related to this device.

### Which version of ESPHome has the issue?

2025.6.3

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2025.7.0

### What platform are you using?

ESP8266

### Board

Sonoff TH10/TH16

### Component causing the issue

dht / si7021

### YAML Config

```yaml
sensor:
  - platform: dht
    model: Si7021
    pin: GPIO14
    humidity:
      name: "Humidity"
    temperature:
      name: "Temperature"
    update_interval: 60s
```

### Anything in the logs that might be useful for us?

```txt
[dht:051]: Invalid readings! Check pin number and pull-up resistor and try manually specifying the model.
```

Tadski, 2025-07-20:
Identical issue after upgrading from 2025.6.x to 2025.07.2, similar config to OP.

picanl, 2025-09-10:
Esphome 2025.8.4 fixes the issue.