esphome/issues #7207: After updating from ESPHome 2025.4.2 to newer versions (2025.6), my previously stable project stopped working reliably.

Git-HAndy, 2025-07-06:
**ESPHome version:**
- Working: 2025.4.2
- Broken: 2025.6.x

**Board:**
- Inkplate 10.1 (ESP32)

**Problem Summary:**
After updating from ESPHome 2025.4.2 to newer versions (2025.6), my previously stable project stopped working reliably.

Symptoms:
- ESP connects via OTA, but API connection hangs for 30–60s or fails completely
- After reboot, the display remains stuck or throws `component display is marked FAILED`
- MCP23017 and PCA6416A-related warnings appear even though unused components were removed or properly configured
- Display often stays on the startup screen and doesn't progress
- Overall, startup time and API stability degraded severely

**What works:**
- Downgrading back to ESPHome 2025.4.2 immediately restores expected behavior
- Display initializes normally and HA sensors are populated within seconds

**Heap memory observation:**
- Free heap (measured via `heap_caps_get_free_size(MALLOC_CAP_INTERNAL)`) dropped significantly under 2025.6, often below 200,000 bytes
- Under 2025.4.2, free heap stays above 215,000–225,000 bytes during normal operation
- This may indicate a memory usage increase or leak in recent versions

**Suspected cause:**
- Possibly a RAM/resource timing issue due to increased memory use in newer versions
- Display component possibly initializes before required I²C devices are ready

**Expected behavior:**
- ESP boots cleanly and quickly
- Display updates within seconds
- API connection is fast and stable

**Minimal working example:**
(I can provide the full YAML file if needed. The configuration is too large to paste inline.)

Would appreciate help in tracing what changed between 2025.4.x and later. Let me know if I can help with testing or debugging. Thank you!

swoboda1337, 2025-07-07:
Can you try to set cpu_frequency to 240: https://esphome.io/components/esp32.html#configuration-variables

Git-HAndy, 2025-07-07:
Update: I upgraded to ESPHome 2025.6.3 and added `cpu_frequency: 240MHz` as suggested.

So far (after 20+ minutes uptime), the display initialized immediately, API connected on first attempt, and there have been no disconnects or failures. Free heap is around 195 KB, which seems stable.

I'll continue monitoring, but first impression: this seems to have resolved the problem.

Thank you!

For my understanding. Could you briefly describe why it previously worked at 160 MHz and now requires 240 MHz?

swoboda1337, 2025-07-07:
If you were using Arduino the default was 240 previously but was changed to 160 in a recent build