esphome/issues #7210: toggling cover from home assisant 2025.7.1 makes esp restart

kvanbiesen, 2025-07-06:
### The problem

I have home assistant 2025.7.1 running and it messed up the cover template.
When i roll back everything is fine.

i have a shelly, that acts as a garage door opener. I use the toggle switch in HA (2025.7.1), the device does some kinda soft restart. Most of the time happens after some idle time.  On the 2025.6.3, it never happens.

Nothing shows in the HA logs, only that the device is not reachable.

Ive run an esp log and have it attached

### Which version of ESPHome has the issue?

2025.6.3

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2027.7.1

### What platform are you using?

ESP8266

### Board

esp01_1m (shelly 1)

### Component causing the issue

cover i think

### YAML Config

```yaml
substitutions:
  friendly_name: Garage

esphome:
  name: shelly-garage

esp8266:
  board: esp01_1m

<<: !include .base.yaml
debug:
  update_interval: 5s
# The door contact sensor that is attached to SW on the 
# Shelly 1. Not exposed to HA, instead used to set the 
# state of the cover we.
binary_sensor:
  - platform: gpio
    pin: GPIO5
    name: "Garage Door Contact Sensor"
    id: contact_sensor
    internal: true
    filters:
      - invert:
      # Debounce the contact sensor to prevent rapid on/off/on events
      - delayed_on_off: 100ms

# The relay in the Shelly 1 that will deliver the pulse to
# the garage door opener (not exposed to HA)
switch:
  - platform: gpio
    pin: GPIO4
    name: "Garage Door Relay"
    id: relay
    internal: true

# This creates the actual garage door in HA. The state is based
# on the contact sensor. Opening/closing the garage door simply
# turns the relay on/off with a 0.5s delay in between.
cover:
  - platform: template
    device_class: garage
    name: "Garage Door"
    id: template_cov
    assumed_state: true  # Helps HA cope with ambiguous state logic
    lambda: |-
      if (id(contact_sensor).state) {
        return COVER_OPEN;
      } else {
        return COVER_CLOSED;
      }
    open_action:
      - switch.turn_on: relay
      - delay: 0.9s
      - switch.turn_off: relay
    close_action:
      - switch.turn_on: relay
      - delay: 0.9s
      - switch.turn_off: relay

text_sensor:
  - platform: wifi_info
    ip_address:
      name: $friendly_name IP Address
  - platform: debug
    device:
      name: "Device Info"
    reset_reason:
      name: "Reset Reason"

# Send WiFi signal strength & uptime to HA
sensor:
  - platform: wifi_signal
    name: $friendly_name WiFi Strength
    update_interval: 60s
  - platform: uptime
    name: $friendly_name "Uptime"
```

### Anything in the logs that might be useful for us?

```txt

```

### Additional information

as you can see, it reboots for some reason when using a cover template from 2025.7.1.

It doesnt happen on 2025.6.3 which i m running now. I install 2025.7.1, it imediately happens. 

Keeping the connecting alive with logging or ping doesnt help.
Weird thing after the "reboot" it works for like an hour, after that it reboots again when toggling.

[esphome.log](https://github.com/user-attachments/files/21090008/esphome.log)

bdraco, 2025-07-06:
Please follow the guide at https://esphome.io/guides/troubleshooting.html to get a stack trace.

gorfo66, 2025-07-07:
@kvanbiesen not sure this is the same root cause but mine did not expose any sensor anymore, except the firmware info.
I finally restarted HA and everything went back online.

kvanbiesen, 2025-07-07:
Ran a stack trace for hours. The device restarted for z couple of times.
But not errors. It just restarted.

I rebooted the whole server and for now it seems to work. I'm on 2025.7.1

I'm gonna close this for now. I'll dig deeper into it if it happens again

closed 2025-11-21 by github-actions[bot]