esphome/issues #7211: can't use external components

zqrup, 2025-07-06:
### The problem

i am trying to modify the esp32_camera component to work with my ov7670 module, and to do this i am using the "external_components" component but since i upgraded to 2025.6.3 it just give this error:

```
INFO ESPHome 2025.6.3
INFO Reading configuration /config/esphome/esp32.yaml...
ERROR Unable to import component esp32_camera:
Traceback (most recent call last):
  File "/esphome/esphome/loader.py", line 186, in _lookup_module
    module = importlib.import_module(f"esphome.components.{domain}")
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/importlib/__init__.py", line 90, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1387, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1360, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1331, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 935, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 999, in exec_module
  File "<frozen importlib._bootstrap>", line 488, in _call_with_frames_removed
  File "/data/external_components/ef121e91/esphome/components/esp32_camera/__init__.py", line 22, in <module>
    from esphome.core.entity_helpers import setup_entity
ImportError: cannot import name 'setup_entity' from 'esphome.core.entity_helpers' (/esphome/esphome/core/entity_helpers.py)
Failed config

esp32_camera: [source /config/esphome/esp32.yaml:30]
  
  Component not found: esp32_camera.

```

ive also tried putting the external component files on the home assistant host but it gave the same error.

please tell me if you need more info.

### Which version of ESPHome has the issue?

2025.6.3

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

Core: 2025.7.1, Supervisor: 2025.06.2, Operating System: 15.2

### What platform are you using?

ESP32

### Board

ESP32 devkit C V4

### Component causing the issue

external_components

### YAML Config

```yaml
esphome:
  name: esp32
  friendly_name: esp32

esp32:
  board: esp32dev
  framework:
    type: esp-idf

logger:

api:
  encryption:
    key: ""

ota:
  - platform: esphome
    password: ""

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

external_components:
  - source: github://zqrup/esphome@dev
    components: [ esp32_camera ]

esp32_camera:
  name: espcam
  i2c_pins: 
    scl: GPIO22
    sda: GPIO21
  external_clock: 
    pin: GPIO0
    frequency: 20MHz
  pixel_clock_pin: GPIO27
  vsync_pin: GPIO25
  href_pin: GPIO23
  data_pins: [GPIO36, GPIO39, GPIO34, GPIO35, GPIO18, GPIO19, GPIO4, GPIO5]
  reset_pin: GPIO33
  power_down_pin: GPIO2
```

### Anything in the logs that might be useful for us?

```txt
INFO ESPHome 2025.6.3
INFO Reading configuration /config/esphome/esp32.yaml...
ERROR Unable to import component esp32_camera:
Traceback (most recent call last):
  File "/esphome/esphome/loader.py", line 186, in _lookup_module
    module = importlib.import_module(f"esphome.components.{domain}")
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/importlib/__init__.py", line 90, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1387, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1360, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1331, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 935, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 999, in exec_module
  File "<frozen importlib._bootstrap>", line 488, in _call_with_frames_removed
  File "/data/external_components/ef121e91/esphome/components/esp32_camera/__init__.py", line 22, in <module>
    from esphome.core.entity_helpers import setup_entity
ImportError: cannot import name 'setup_entity' from 'esphome.core.entity_helpers' (/esphome/esphome/core/entity_helpers.py)
Failed config

esp32_camera: [source /config/esphome/esp32.yaml:30]
  
  Component not found: esp32_camera.
```

ssieb, 2025-07-06:
There's something wrong with the external component code.

guybogaarts, 2025-07-07:
I have the same error, using version 2025.6.0

ssieb, 2025-07-07:
@guybogaarts what is your config and what is the error?  This issue is specifically about an external component.

closed 2025-07-20 by zqrup