esphome/issues #7213: esp32-c6-devkitc-1 use ESPHome 2025.6.0 Unable to start

KiritakeKumi, 2025-07-07:
### The problem

esp32-c6-devkitc-1 may have a problem with the default configuration.

According to my investigation, it seems to be a problem with ESP-IDF. I replaced the 5.2.1 version and it was able to start and run normally.

And I found that using the default ESP-IDF version of ESPHome 2025.6.0, the configuration information is as follows:
HARDWARE: ESP32C6 160MHz, 320KB RAM, 8MB Flash

Using 5.2.1 is like this:
HARDWARE: ESP32C6 160MHz, 512KB RAM, 8MB Flash

It seems to be a problem caused by memory capacity error?

I will continue to investigate the ESP-IDF configuration issue and will continue to update, but I would like to ask if there are users who have the same experience.

### Which version of ESPHome has the issue?

2025.6

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP32

### Board

esp32-c6-devkitc-1

### YAML Config

```yaml

```

### Anything in the logs that might be useful for us?

```txt

```

swoboda1337, 2025-07-07:
Did you try 2025.6.2? There was a workaround for an I2C issue

KiritakeKumi, 2025-07-09:
> Did you try 2025.6.3? There was a workaround for an I2C issue

Doesn’t seem to be a problem?

My current error log is as follows:

[17:46:49]ESP-ROM:esp32c6-20220919
[17:46:49]Build:Sep 19 2022
[17:46:49]rst:0x1 (POWERON),boot:0xc (SPI_FAST_FLASH_BOOT)
[17:46:49]SPIWP:0xee
[17:46:49]mode:DIO, clock div:2
[17:46:49]load:0x40875720,len:0x18ec
[17:46:49]load:0x4086c110,len:0xd9c
[17:46:49]load:0x4086e610,len:0x2f80
[17:46:49]entry 0x4086c110

And then nothing happened.