esphome/issues #7232: [libretiny] template number not stored in the flash correctly

szupi-ipuzs, 2025-07-13:
### The problem

This is my experimental code to make a sleeping device prepare for OTA. The idea is to publish a token (externally, eg. via HA) to a specific mqtt topic (retained) and the device needs to check if it has already seen this token - if not, then it prevents sleep for some time.

So I store the token in the update_token_number (template number, internal). But it doesn't seem to be written to the flash correctly. After reboot it contains the initial value instead of the restored value.
The test scenario is simple: publish the token (different than stored) to the mqtt topic, then reboot the device via dedicated reboot button.

Here's the partial log starting with updating the token:

```
[2025-07-13 23:48:31.434] [D][main:059][async_tcp]: Received token 345679, waiting for update...
[2025-07-13 23:48:31.440] [D][number:012][async_tcp]: 'update_token_number': Sending state 345679.000000
[2025-07-13 23:48:31.450] [V][text_sensor:015][async_tcp]: 'Update token': Received new state 345679
[2025-07-13 23:48:31.456] [D][text_sensor:069][async_tcp]: 'Update token': Sending state '345679'
[2025-07-13 23:48:31.467] [V][mqtt:508][async_tcp]: Publish(topic='cbu-dev-board-1/sensor/update_token/state' payload='345679' retain=1 qos=0)
[2025-07-13 23:48:37.555] [V][sensor:051]: 'Uptime': Received new state 18.978001
[2025-07-13 23:48:37.560] [D][sensor:104]: 'Uptime': Sending state 18.97800 s with 0 decimals of accuracy
[2025-07-13 23:48:37.566] [V][mqtt:508]: Publish(topic='cbu-dev-board-1/sensor/uptime/state' payload='19' retain=1 qos=0)
[2025-07-13 23:49:18.635] [I][safe_mode:042]: Boot seems successful; resetting boot loop counter
[2025-07-13 23:49:18.641] [VV][lt.preferences:041]: s_pending_save: key: 233825507, len: 4
[2025-07-13 23:49:18.647] [V][lt.preferences:104]: Saving 1 items...
[2025-07-13 23:49:18.652] [VV][lt.preferences:113]: Checking if FDB data 233825507 has changed
[2025-07-13 23:49:18.658] [V][lt.preferences:115]: sync: key: 233825507, len: 4
[2025-07-13 23:49:18.663] [D][lt.preferences:132]: Writing 1 items: 0 cached, 1 written, 0 failed
[2025-07-13 23:49:18.673] [VV][component:203]: safe_mode loop disabled
[2025-07-13 23:49:37.556] [V][sensor:051]: 'Uptime': Received new state 78.977997
[2025-07-13 23:49:37.561] [D][sensor:104]: 'Uptime': Sending state 78.97800 s with 0 decimals of accuracy
[2025-07-13 23:49:37.566] [V][mqtt:508]: Publish(topic='cbu-dev-board-1/sensor/uptime/state' payload='79' retain=1 qos=0)
[2025-07-13 23:50:31.268] [D][button:010][async_tcp]: 'Restart' Pressed.
[2025-07-13 23:50:31.273] [I][restart.button:012][async_tcp]: Restarting device
[2025-07-13 23:50:31.370] [I][app:198][async_tcp]: Rebooting safely
[2025-07-13 23:50:31.376] [VV][lt.preferences:064][async_tcp]: fdb_kv_get_blob: key: 233825507, len: 4
[2025-07-13 23:50:31.382] [VV][lt.preferences:041][async_tcp]: s_pending_save: key: 233825507, len: 4
[2025-07-13 23:50:31.392] [V][lt.preferences:104][async_tcp]: Saving 1 items...
[2025-07-13 23:50:31.398] [VV][lt.preferences:113][async_tcp]: Checking if FDB data 233825507 has changed
[2025-07-13 23:50:31.409] [D][lt.preferences:127][async_tcp]: FDB data not changed; skipping 233825507  len=4
[2025-07-13 23:50:31.415] [D][lt.preferences:132][async_tcp]: Writing 1 items: 1 cached, 0 written, 0 failed
[2025-07-13 23:50:31.425] [V][mqtt:508][async_tcp]: Publish(topic='cbu-dev-board-1/status' payload='offline' retain=1 qos=0)
```

And here's what happens after reboot
```
[2025-07-13 23:50:31.586] [VV][lt.preferences:064]: fdb_kv_get_blob: key: 233825507, len: 4
[2025-07-13 23:50:31.598] [C][safe_mode:082]: There have been 0 suspected unsuccessful boot attempts
[2025-07-13 23:50:31.603] [VV][lt.preferences:041]: s_pending_save: key: 233825507, len: 4
[2025-07-13 23:50:31.609] [V][lt.preferences:104]: Saving 1 items...
[2025-07-13 23:50:31.614] [VV][lt.preferences:113]: Checking if FDB data 233825507 has changed
[2025-07-13 23:50:31.620] [V][lt.preferences:115]: sync: key: 233825507, len: 4
[2025-07-13 23:50:31.625] [D][lt.preferences:132]: Writing 1 items: 0 cached, 1 written, 0 failed
[2025-07-13 23:50:31.637] [I][app:049]: Running through setup()
[2025-07-13 23:50:31.637] [V][app:050]: Sorting components by setup priority
[2025-07-13 23:50:31.642] [VV][component:203]: preferences loop disabled
[2025-07-13 23:50:31.691] [VV][lt.preferences:061]: NVS length does not match (0!=4)
[2025-07-13 23:50:31.696] [D][number:012]: 'update_token_number': Sending state 345678.000000
```

I'm using the yaml on my cbu devboard and logging via serial. I have my external component for bk72xx deepsleep included. Still the issue is not with sleeping but with rebooting so I don't think this adds to the problem.

### Which version of ESPHome has the issue?

2025.8-dev

### What type of installation are you using?

Docker

### What platform are you using?

BK72XX

### Board

tuya cbu devboard

### Component causing the issue

libretiny

### YAML Config

```yaml
substitutions:
  name: cbu-dev-board-1
  friendly_name: CBU Dev Board 1
  ap_password: !secret fallback_wifi_password
  wifi_ssid1: !secret wifi_ssid4
  wifi_password1: !secret wifi_password4
  wifi_ssid2: !secret wifi_ssid2
  wifi_password2: !secret wifi_password2
  wifi_ssid3: !secret wifi_ssid3
  wifi_password3: !secret wifi_password3

packages:
  wifi: !include common/wifi_no_ip.yaml
  device_base: !include common/device_base_mqtt.yaml
  web_server: !include common/web_server.yaml

external_components:
  - source:
      type: local
      path: /my-external-components/components

deep_sleep_libretiny:
  id: deep_sleep_lt
  run_duration: 30s
  sleep_duration: 10s

logger:
  level: VERY_VERBOSE

esphome:
  platformio_options:
    build_flags:
      - "-D LWIP_DHCP_DOES_ACD_CHECK=0"
      - "-D LT_UART_DEFAULT_PORT=2"
  on_boot:
    - priority: 500
      then:
        - text_sensor.template.publish:
            id: update_token_number_text
            state: !lambda 'return std::to_string(id(update_token_number).state);'

bk72xx:
  board: cbu
  framework:
    version: 0.0.0
    source: "/libretiny"

mqtt:
  broker: 192.168.2.254
  discovery: True
  log_topic: null
  on_message:
    - topic: ${name}/ota_mode
      then:
        - lambda: |-
            auto token_from_mqtt = atoi(x.c_str());
            if ((token_from_mqtt == 0) || (token_from_mqtt > id(update_token_number).state))
            {
              ESP_LOGD("main", "Received token %d, waiting for update...", token_from_mqtt);
              if (token_from_mqtt == 0)
              {
                token_from_mqtt = id(update_token_number).state + 1;
              }

              id(update_token_number).publish_state(token_from_mqtt);
              id(update_token_number_text).publish_state(std::to_string(token_from_mqtt));
              if (!id(wait_for_update_script).is_running())
              {
                id(wait_for_update_script).execute();
              }
            }
            else
            {
              ESP_LOGD("main", "Received token %d, but current token is %d, ignoring...", token_from_mqtt, id(update_token_number).state);
            }

number:
  - platform: template
    id: update_token_number
    update_interval: never
    min_value: 0
    max_value: 1000
    step: 1
    optimistic: True
    initial_value: 345678
    internal: True
    restore_value: True

text_sensor:
  - platform: template
    id: update_token_number_text
    name: "Update token"
    update_interval: never

script:
  - id: wait_for_update_script
    mode: single
    then:
      - deep_sleep_libretiny.prevent: deep_sleep_lt
      - delay: 10min
      - deep_sleep_libretiny.allow: deep_sleep_lt
```

### Anything in the logs that might be useful for us?

```txt

```

ssieb, 2025-07-13:
`publish_state` is not the right way to change a number, so it's not going to save it.  You need to use a call as described in the docs.  The number you're setting is also outside the range you've specified, so that might also interfere.

closed 2025-07-13 by ssieb

szupi-ipuzs, 2025-07-14:
OK, I assume you're refering to the number.set action, but I did not find its counterpart that would be callable from a lambda. Could you tell me which one it is?

On the side note: shouldn't it be impossible to set a value outside the [mix, max] range by any of the calls? As you could see it succeeded. Should I file a separate bug for that?